SpikeInterface/spikeinterface #3390: Local Kilosort3 Does not Save Output with run_sorter()

masahirotaki, 2024-09-10:
Hi,

The docker version of kilosort 3 gives me an issue with CUDA error on one of my recordings (seems common and adding try and catch works in the local version).

So I decided to run spikeinterface with local kilosort3 for this particular recording:

`si.Kilosort3Sorter.set_kilosort3_path('/home/saleem_lab/Kilosort')`

` probe0_preprocessed_corrected = si.load_extractor(save_folder+'probe'+str(probe)+'_preprocessed')`
`probe0_sorting_ks3 = si.run_sorter(sorter_name='kilosort3',recording=probe0_preprocessed_corrected,output_folder=save_folder+'probe'+str(probe)+'/sorters/kilosort3/',do_correction=False)`

Everything runs smoothly with it and it produces `spikeinterface_log.json` which shows that the sorting ran successfully.

But the sorter_output folder is empty.

I also ran sorting_analyzer on the sorting and extracted waveforms successfully.

In the sorting_analyzer folder, `~/sorting/ `contains `si_folder.json` which shows that `"kwargs": {
        "folder_path": "."
    },
    "annotations": {
        "phy_folder": "/home/saleem_lab/spikeinterface_sorting/temp_data/20240606/probe1/sorters/kilosort3/sorter_output",
        "__sorting_info__": {`.

So I'm really confused why the `sorter_output` folder is empty...

zm711, 2024-09-10:
could you print(probe0_sorting_ks3.unit_ids)?

If the folder is empty I don't know how you would have units and without units I don't know how you would be able to make a sorting_analyzer? If you could post the spikeinterface_log.json here we could also double check it.

And is this folder completely empty? If Kilosort fails it usually has some .mat files within the folder.

zm711, 2024-09-10:
Also we deprecated output_folder in favor of just folder. could you try just feeding in folder instead? Maybe we have a bug with the deprecated argument. Which version of spikeinterface are you using?

masahirotaki, 2024-09-12:
Thank you so much for your reply!
We are running spikeinterface 0.101.0.
For the purpose of analysis pipeline, for now, it will be great if we can keep output_folder. But we will consider the option if it is necessary.

I just had the chance to run everything step by step using ipython. I noticed that after running 
`probe0_sorting_ks3` = `si.run_sorter(sorter_name='kilosort3',recording=probe0_preprocessed_corrected,output_folder=save_folder+'probe'+str(probe)+'/sorters/kilosort3/',do_correction=False)`

It is interesting that I don't see in_container_sorter folder. But the sorter_output folder looks fine at this stage! It might got deleted afterwards by acccident? Currently, running line by line to confirm this. 

[image]

And this is the output of `print(probe0_sorting_ks3.unit_ids)`：
[  0   1   2   3   4   5   6   7   8   9  10  11  12  13  14  15  17  18
  19  21  22  23  24  25  29  30  31  33  34  35  36  37  38  39  40  41
  42  43  45  46  47  49  50  51  52  53  54  55  56  57  58  59  60  61
  62  63  64  65  66  67  68  69  72  74  75  76  80  81  82  84  85  86
  89  90  92  93  94  95  96  97  98  99 100 101 102 103 104 105 106 107
 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125
 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 142 143 144
 145 146 147 148 149 150 151 152 154 155 157 158 159 160 162 163 164 169
 172 173 174 175 177 178 181 183 187 188 189 190 191 193 194 195 196 197
 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215
 216 217 218 219 220 221 222 223 224 225 226 227 228 230 232 233 234 236
 237 238 239 243 244 245 246 247 248 249 250 251 252 253 254 255 256 257
 258 259 260 261 262 263 264 265 266 267 268 269 270 271 272 274 275 277
 278 279 280 281 284 285 287 289 291 292 293 294 295 296 297 298 299 300
 302 303 304 305 306 307 308 310 311 312 313 314 315 316 317 318 319 321
 322 323 324 325 326 327 329 330 331 332 333 334 335 336 337 338 339 340
 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358
 359 360 361 362 363 364 365 366 367 368 369 370 371 373 374 375 377 378
 379 380 381 382 383 384 385 386 387 388 389 390 391 392 393 394 395 396
 397 398 399 400 401 402 403 404 406 407 410 411 412 413 414 415 416 417
 418 419 420 421 422 423 424 425 426 427 428 430 431 432 433 434 435 436
 438 439 440 441 442 443 444 445 446 447 448 449 450 451 452 453 454 456
 459 460 462 463 464 465 466 468 469 470 471 472 473 474 475 476 477 478
 479 480 482 483 484 485 486 487 488 489 490 491 492 493 494 495 496 497
 498 499 500]

zm711, 2024-11-22:
@masahirotaki, did you figure this out? I was waiting for you to see when the folder was getting deleted?

masahirotaki, 2024-11-26:
Hi Zach,

Sorry for the late response. We were so busy with other stuff, that we didn't have time to test too much. Basically, a combination of reinstallation and cleaning up some codes and using local KS3 without using container worked!... But unfortunately, we didn't know the exact cause for sure. The output folder was probably deleted later? Basically, by the time I am testing line by line, it was working. Sorry it is not that informative.

closed 2024-11-26 by alejoe91