SpikeInterface/spikeinterface #3394: Exporting to phy changes cluster channels

RikkelBob, 2024-09-11:
After manual curation, I'm recalculating cluster waveforms and some quality metrics using the script below. However, when I export to phy, the channel numbers are wrong. When I pause/debug the script before `export_to_phy`, channel numbers are correct, but inspecting cluster_info.tsv after exporting, they're different. I would fix this post-hoc, but because the channel numbers are wrong, the calculated waveforms and subsequent waveform based quality metrics are also incorrect. I'd very much appreciate any help!

```
                  rec_folder = Path(probe_folder, probe)
                        probe_number = find_numbers_in_string(probe)
                        phy_dir = rec_folder
                        ac_dir = Path(phy_dir, 'postprocessed')
                        ac_dir_wave = Path(ac_dir, 'waveforms')
                        ac_dir_phy = Path(ac_dir, 'phy')

                        if not os.path.isdir(ac_dir_phy):

                            waveform_folder = Path(phy_dir, "analyzer")
                            report_folder = Path(phy_dir, 'report')

                            # Load your sorted units from the Phy folder
                            sorting = read_phy(phy_dir)

                            # Get channel locations from .npy file
                            path_channel_location = Path(phy_dir, 'channel_positions.npy')
                            channel_locations = np.load(path_channel_location)
                            num_channels = channel_locations.shape[0]

                            # Load the original recording
                            recording = si.read_binary(Path(ses_folder, "ksData_probe" + str(probe_number[0]) + ".dat"),
                                                       sampling_frequency=fs,
                                                       num_channels=num_channels,
                                                       dtype="int16")

                            probe_64, locations = utils.gen_probe()
                            recording.set_probe(probe_64, in_place=True)

                            #recording.set_dummy_probe_from_locations(channel_locations)
                            recording.annotate(is_filtered=True)

                            sorting = remove_excess_spikes(sorting, recording)
                            sorting = remove_duplicated_spikes(sorting=sorting, censored_period_ms=0.5, method="keep_first")
                            #sorting = sorting.remove_empty_units()

                            print("building analyzer")
                            analyzer = si.create_sorting_analyzer(sorting=sorting, recording=recording, return_scaled=False)

                            analyzer.compute("random_spikes")
                            analyzer.compute("waveforms", save=True)
                            templates = analyzer.compute("templates", save=True)
                            analyzer.compute(
                                "spike_amplitudes",
                                peak_sign="pos"
                            )
                            analyzer.compute("noise_levels")

                            amp_cutoff = analyzer.compute(
                                "quality_metrics",
                                qm_params=dqm_params
                            )

                            print("Saving analyizer")
                            analyzer.save_as(format="binary_folder", folder=ac_dir_wave)
                            print("Exproting to phy")
                            export_to_phy(analyzer, output_folder=ac_dir_phy, copy_binary=False, additional_properties=['quality'], remove_if_exists=True)
                            print("Finished" + str(ac_dir_phy))

```

zm711, 2024-09-11:
Maybe related #3392? I'll have to read a bit more later. Are you using a probe group or probe?

RikkelBob, 2024-09-11:
@zm711 I don't use probe groups, just a probe - `recording.set_probe(probe_64, in_place=True)`

zm711, 2024-09-11:
Internally we representing probes as a probegroup (see here https://github.com/SpikeInterface/spikeinterface/blob/73f4d58d2d542e607155d49d30716cc28215e12f/src/spikeinterface/core/sortinganalyzer.py#L1101-L1106) so if there is some sort of bug happening when we get the probegroup (to get your probe) it could affect this as well. Do you want to try exporting from that PR and see if it fixes things or if it is still broken.

RikkelBob, 2024-09-11:
I've tried it now, but the issue persists

zm711, 2024-09-11:
Okay thanks. We will read this more closely soon and see if we can find the place soon.

RikkelBob, 2024-09-12:
I suspect the issue to be within the `export_to_phy` function. Both the loaded sorting object and the analyzer object describe the correct channels.

RikkelBob, 2024-10-15:
Would love to hear it if anything has been found related to this :)

samuelgarcia, 2024-10-16:
@alejoe91 could you check this?

chiyu1203, 2025-04-17:
Hi @samuelgarcia @alejoe91 @RikkelBob ,
I just bumped into this issue (and maybe related to #3563). I was doing manual curation on phy and found out that clusters of the same **channel_group** are not marked at the same shank on the **ProbeView**. 

[image]

This was odd to me. I checked the probe map before and did not think that I have messed up the map. And I followed the tutorial to group the channels by shank and analysed them shank by shank during common reference, motion correction, whitening, spike sorting. During postprocessing step, I then aggregated the groups before creating the sorting_analyser. Is there anything might go wrong during these steps? 
Below is the minimum version of my code (spikeinterface version **0.102.2**, built from source) and the probe map. Feel free to let me know if you have any thought.

```
raw_rec = se.read_openephys(oe_folder, load_sync_timestamps=True)
stacked_probes = pi.read_probeinterface("H10_stacked_probes_2D.json")
probe = stacked_probes.probes[0]
raw_rec = raw_rec.set_probe(probe,group_mode='by_shank')
recording_f = spre.bandpass_filter(raw_rec, freq_min=600, freq_max=6000)
bad_channel_ids,channel_labels = spre.detect_bad_channels(recording_f, method="coherence+psd")
recording_f = recording_f.remove_channels(bad_channel_ids)
recordings_dict = recording_f.split_by(property='group', outputs='dict')
recording_cmr = spre.common_reference(recordings_dict, reference="global", operator="median")
recording_corrected_dict=motion_correction_shankbyshank(recording_saved,oe_folder,analysis_methods)
rec_for_sorting = spre.whiten(
                recording=recording_corrected,
                mode="local",
                radius_um=150)
rec_for_sorting=si.aggregate_channels(rec_for_sorting)
sorting_spikes = ss.run_sorter_by_property(
                sorter_name=this_sorter,
                recording=rec_for_sorting,
                remove_existing_folder=True,
                grouping_property='group',
                folder=oe_folder / result_folder_name,
                verbose=True,
                **sorter_params
                )
sorting_analyzer = si.create_sorting_analyzer(
            sorting=sorting_spikes,
            recording=recording_for_analysis,
            sparse=True,  # default
            format="memory",  # default)
calculate_analyzer_extension(sorting_analyzer)
sep.export_to_phy(
                sorting_analyzer,
                output_folder=oe_folder / phy_folder_name,
                compute_amplitudes=True,
                compute_pc_features=True,
                copy_binary=False,
                remove_if_exists=True)
```
[H10_stacked_probes_2D.json](https://github.com/user-attachments/files/19798509/H10_stacked_probes_2D.json)