SpikeInterface/spikeinterface #3397: Kilosort4 being passed wrong information in run_sorter_by_property

mdmornin, 2024-09-11:
I am revamping my spikesorting workflow to individually sort shanks. I have been able to do all the requisite steps, code included below:

When attempting to run Kilosort4, the following traceback is called:
```
========================================
Loading recording with SpikeInterface...
number of samples: 143981568
number of channels: 8
numbef of segments: 1
sampling rate: 20000.0
dtype: int16
========================================
Preprocessing filters computed in  82.52s; total  82.52s

computing drift
Re-computing universal templates from data.
  0%|          | 0/2400 [00:00<?, ?it/s]Error running kilosort4

Traceback (most recent call last):

  Cell In[1], line 68
    aggregate_sorting = ss.run_sorter_by_property(sorter_name='kilosort4',

  File ~\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\launcher.py:306 in run_sorter_by_property
    sorting_list = run_sorter_jobs(job_list, engine=engine, engine_kwargs=engine_kwargs, return_output=True)

  File ~\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\launcher.py:106 in run_sorter_jobs
    sorting = run_sorter(**kwargs)

  File ~\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\runsorter.py:216 in run_sorter
    return run_sorter_local(**common_kwargs)

  File ~\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\runsorter.py:276 in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)

  File ~\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\basesorter.py:301 in run_from_folder
    raise SpikeSortingError(

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\external\kilosort4.py", line 256, in _run_from_folder
    ops, bfile, st0 = compute_drift_correction(
  File "C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\kilosort\run_kilosort.py", line 344, in compute_drift_correction
    ops, st = datashift.run(ops, bfile, device=device, progress_bar=progress_bar)
  File "C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\kilosort\datashift.py", line 192, in run
    st, _, ops  = spikedetect.run(ops, bfile, device=device, progress_bar=progress_bar)
  File "C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\kilosort\spikedetect.py", line 243, in run
    tF[k:k+nsp] = xfeat.transpose(0,1).cpu().numpy()
ValueError: could not broadcast input array from shape (252,8,6) into shape (252,10,6)

Spike sorting failed. You can inspect the runtime trace in C:\Users\Mitch\test\0/spikeinterface_log.json.
```
The key message being:
`ValueError: could not broadcast input array from shape (252,8,6) into shape (252,10,6)`

Running get_traces and get_probe produce the correct shape (8 channels). Running mountainsort5 has so far worked leading me to believe there may be an issue with how the wrapper is handling the grouped information?

The full code minus imports is below:
`recording_file = r'ID-HDR3_Day-16_Epoch-1_numCh-64_probe-F_headstage-miniAmp_sessionType-cueSwap_09-Sep-2024-12-20-18_ephys.dat'
num_channels = 64
sampling_frequency = 20000
   
dtype = 'uint16'
sorter = 'kilosort4'
gain_to_uV = 0.195
offset_to_uV = 32768

recording = si.read_binary(recording_file, 
                           num_channels=num_channels, 
                           sampling_frequency=sampling_frequency,
                           dtype=dtype,
                           gain_to_uV=gain_to_uV,
                           offset_to_uV=offset_to_uV)

manufacturer = 'cambridgeneurotech'
probe_name = 'ASSY-236-F'
probe = get_probe(manufacturer,probe_name)
plot_probe(probe, with_contact_id=True)
probe.wiring_to_device('cambridgeneurotech_mini-amp-64')
plot_probe(probe, with_contact_id=True)

       

recording = recording.set_probe(probe,group_mode='by_shank')

preprocessed_recordings = []
badCh = []
split_recording_dict = recording.split_by('group')
for chan_group_rec in split_recording_dict.values():
    cleanRec = spre.bandpass_filter(recording=chan_group_rec, freq_min=300, freq_max=6000)
    badCHIDs, chLabels = spre.detect_bad_channels(recording=cleanRec,method='coherence+psd');
    cleanRec = cleanRec.remove_channels(remove_channel_ids=badCHIDs)
    print(chLabels)
    cleanRec = spre.common_reference(recording=cleanRec)
    preprocessed_recordings.append(cleanRec)
    badCh.append(badCHIDs)
    
cleanRecComb = si.aggregate_channels(preprocessed_recordings)

#params_sort = {'dminx' : 400, 'nearest_templates' : 50, 'nblocks' : 0}

aggregate_sorting = ss.run_sorter_by_property(sorter_name='mountainsort5', 
                    recording=cleanRecComb, 
                    folder='test', 
                    verbose=True,
                    grouping_property='group',
                    remove_existing_folder=True)
`

environment information:
`
     active environment : HDR
    active env location : C:\Users\Mitch\.conda\envs\HDR
            shell level : 2
       user config file : C:\Users\Mitch\.condarc
 populated config files :
          conda version : 24.1.2
    conda-build version : 24.1.2
         python version : 3.11.7.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=x86_64
                          __conda=24.1.2=0
                          __cuda=12.5=0
                          __win=0=0
       base environment : C:\ProgramData\anaconda3  (read only)
      conda av data dir : C:\ProgramData\anaconda3\etc\conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\ProgramData\anaconda3\pkgs
                          C:\Users\Mitch\.conda\pkgs
                          C:\Users\Mitch\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Mitch\.conda\envs
                          C:\ProgramData\anaconda3\envs
                          C:\Users\Mitch\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/24.1.2 requests/2.31.0 CPython/3.11.7 Windows/10 Windows/10.0.22631 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.6 aau/0.4.3 c/LQqsKndfB9ncQ_IEMEVUEA s/msw9c-LROP7KLSZPZP7zHw e/SlVwHnue7S_x3fPmdri2eg
          administrator : False
             netrc file : None
           offline mode : False
`

It's entirely possible I am missing a step somewhere, but maybe that's informative.

zm711, 2024-09-11:
Maybe something with bad channels is messing with this for the KS wrapper: @chrishalcrow , @JoeZiminski , and @alejoe91 recently updated the wrapper, but it won't appear until the next release (0.101.1). so if you want to install from main you could test that.

Which version of KS4 are you using. It is listed as 4.0 in your list, but that doesn't really help us. Could you do
```python
import kilosort
print(kilosort.__version__)
```
The most recent is 4.0.17 and the one we will support with 0.101.1 (to be released soon) will be at least up to 4.0.16.

zm711, 2024-09-11:
Unfortunately with Kilosort there have been many changes and many bug fixes. So our wrapper only supports certain version of Kilosort4. It is very hard to make sure everything is in sync so our new policy is just to support the most up-to-date kilosort until the codebase over there stabilizes. MS5 is more stable and it's wrapper is stable. If you are comfortable with it you could try to update to KS4==4.0.16 and do spikeinterface from main or wait around a week for us to update and get spikeinterface 0.101.1 (which will only support 4.0.16). If this is really a cuda/pytorch issue on your computer and you can only use 4.0 (which I believe has bugs in it that the Kilosort team has fixed) then you would have to use it outside of the spikeinterface wrapper.

chrishalcrow, 2024-09-11:
Hi @mdmornin , I've had problems with kilosort when using recordings with less than 10 channels, so this your problems might be related to that. I've not looked into it properly yet - just flagging it while I remember!

mdmornin, 2024-09-11:
Hi Chris,
I had some of those issues before with a 16ch electrode, but was able to solve them by skipping drift correction and passing the following parameters: 
`params_sort = {'dminx' : 400, 'nearest_templates' : 50, 'nblocks' : 0}`

I upgraded KS4 to 4.0.16, 4.0.14, and 4.0.17 and received the following traceback:
```

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\spikeinterface\sorters\external\kilosort4.py", line 208, in _run_from_folder
    filename, data_dir, results_dir, probe = set_files(settings, filename, probe, probe_name, data_dir, results_dir)
TypeError: set_files() missing 1 required positional argument: 'bad_channels'

Spike sorting failed. You can inspect the runtime trace in C:\Users\Mitch\test\0/spikeinterface_log.json.
```

I am assuming this is because of the missing, new wrapper based on some of the comments I saw in #3339. I attempted to build from Main with the setup.py file in my existing environment but it didn't change the traceback. It's possible I didn't run setup.py correctly as well, I haven't ran a setup.py file in a while so I will try again tomorrow.

JoeZiminski, 2024-09-12:
Hi @mdmornin, indeed I think bad channels was added to the wrapper in #3339 and should be included on the latest version on main. 

Could you do `pip show spikeinterface --version` to see the version. You should get `0.101.0rc1` if the install-from-main worked correctly. Otherwise you can change directly to the top level `spikeinterface` folder your downloaded and try:

```
pip uninstall spikeinterface
pip install -e .
```
this will give you an 'editable' install so if you change the code in your downloaded repo it will be reflected in your installed, imported spikeinterface. Spikeinterface uses the `pyproject.toml` now to coordinate the install as far as I know, so maybe the most recent version was not installing properly.

It will be interesting to see whether this resolves the problem. If not, a couple of potential causes:
- Like Zach said maybe there is something strange in how the spikeinterface wrapper in KS4 is handling things when bad channels are removed. There are workarounds, but could you try removing the bad channel removal and seeing if it runs?
- maybe there is something specifically in the `run_sorter_by_property` wrapper, although I doubt it as it is quite lightweight.

mdmornin, 2024-09-12:
Running `pip show spikeinterface --version' returned 0.101.0. I followed the provided instructions and reinstalled, it then returned 0.101.1. 
I installed originally with: `python setup.py` so having the .toml handled with pip probably fixed it. 

Rerunning the code as is provided the following warning, but so far it seems to be working now with 0.101.1 and KS4 4.0.17. 

```
C:\Users\Mitch\.conda\envs\HDR\lib\site-packages\kilosort\run_kilosort.py:345: UserWarning: 
            Parameter `nearest_chans` must be less than or equal to the number 
            of data channels being sorted.

            Changing from 10 to 8.
            
  warnings.warn(msg, UserWarning)
INFO:kilosort.run_kilosort: 
INFO:kilosort.run_kilosort:Computing preprocessing variables.
INFO:kilosort.run_kilosort:----------------------------------------
INFO:kilosort.run_kilosort:N samples: 143981568
INFO:kilosort.run_kilosort:N seconds: 7199.0784
INFO:kilosort.run_kilosort:N batches: 2400
INFO:kilosort.run_kilosort:Preprocessing filters computed in  87.20s; total  87.20s
INFO:kilosort.run_kilosort: 
INFO:kilosort.run_kilosort:Computing drift correction.
INFO:kilosort.run_kilosort:----------------------------------------
INFO:kilosort.spikedetect:Re-computing universal templates from data.
```

If something weird happens in the results, I'll report back, but so far at least this issue has been cleared up. Thanks all.

JoeZiminski, 2024-09-13:
Great glad to hear it's working @mdmornin, I'll close this for now but please feel free to reopen if any problems arise.

closed 2024-09-13 by JoeZiminski