SpikeInterface/spikeinterface #3399: relationship between electrode and neuron network bursts after sorting

sofiapuvogelvittini, 2024-09-12:
Hello,

I am using SpikeInterface with Spyking Circus 2 to sort Maxwell 2 data. It’s great to observe network bursts (synchronized activity) at both the electrode and sorted-neuronal levels! However,  it seems that after sorting, I consistently "miss" two synchronized events every aprox 50s, which are detected at the electrode level but not in the sorted data. Is this normal? Could you help me interpret or understand what might be causing this?

Attached are plots showing the electrode- and neuron-level raster plots.

Thank you very much for your guidance and for providing such great software!

[Rasterplots.pdf](https://github.com/user-attachments/files/16980786/Rasterplots.pdf)

alejoe91, 2024-09-12:
Hi @sofiapuvogelvittini 

Is it possible that this is a timing misalignment issue? E.g., if the spike interface rasters go beyond 300, then you see all the bursts?

sofiapuvogelvittini, 2024-09-12:
Hi Alession, thanks for your reply. I do not think so, because the length of recording is 300 sec

alejoe91, 2024-09-12:
How are you creating the reatermap on the right?

alejoe91, 2024-09-12:
So you're not using the spikeinterface plot_rasters? Can you try just `sw.plot_rasters(sorting)`?

sofiapuvogelvittini, 2024-09-12:
Attached the plot with the spikeinterface function. But it looks the same.
I first run all computations on a more powerful machine and then download only the output to my laptop for further analysis. That’s why I was using own functions locally.

[image]

alejoe91, 2024-09-12:
Ok, thanks for checking. I have another potential explanation, but it might be a bit far fetched.

Whe you get a very strong and synchronous burst, many many spike swill happen at the same time. If your culture is dense, these spikes will be extremely hard to sort, since the electrodes will record a massive superposition of signal coming from many nearby neurons.. This can be especially true if you chose a network recording!

sofiapuvogelvittini, 2024-09-13:
Oki, in that case, Could you recommend some settings I could adjust to help overcome this issue? Or perhaps suggest a specific sorter that would be most suitable for this scenario? I am currently using SpyKING CIRCUS2.
Thanks Alessio!

yger, 2024-09-13:
Just jumping in the discussion, would you mind sharing a small recording with me? I'm not sure to understand from the raster plots what do you miss exactly, every 50s...

yger, 2024-09-13:
Since I've developed circus2, if there is a parameter to optimize I can tell you, but I need to understand first the issue

sofiapuvogelvittini, 2024-09-13:
Thank you very much for your assistance!

I am trying to understand the relationship between synchronized activity observed at the electrode level and at the sorted neuronal level. From the raster plots at the electrode level, I observe around 35 synchronized events, while at the sorted neuronal level, I can identify only 22 synchronized events... therefore I am not sure if the sorting is missing some events or how can I explain this. 

I have uploaded the recording to the following Google Drive folder: https://drive.google.com/drive/folders/1CoqJTnrn4_qIQEU_rZEc3MuNhaxgHln3?usp=sharing

This dataset is from Maxwell Two, and it contains data from six wells. I am particularly interested in well 0000, which corresponds to rec0000.

The command to read this specific recording should be:
full_raw_rec = se.read_maxwell(h5_data, rec_name='rec0000', stream_id='0000')

For generating the mapping file, you need to use "generate_mapping.py" script (also available in the Google Drive folder). Instructions for using this script are provided in the How_to_use_generate_mapping.txt file. The data.raw.parameters file should be  located in the same folder as the raw data.

Thank you very much for taking the time to look into this!

yger, 2024-09-16:
I'm downloading the data and will have a look. But just to be clear: you are analyzing the exact same data at the spiking/electrode level? i.e including whitening, CAR removal and filtering? Just to rule out the fact that the extra burst might be due to preprocessing. SC2 should not "miss" too many spikes, at least, not regularly every 50s, if the activity is not too synchronous. As @alessio said however, if really it is too synchronous, then sorting might be doomed. How do you analyze your electrode burst? Can you tell me where exactly should I look in the recording to assess if something is going wrong?

sofiapuvogelvittini, 2024-09-16:
Yes, this is exactly the same dataset. The electrode-level analysis is the output of the Maxwell software network scan. However, I am not entirely sure how the data is analyzed or how the spikes are identified. Afterward, I used the .h5 file to perform the sorting. I have included in the gmail drive the script I made to sort across all the wells in the plate (SC2_all_wells_this.ipynb). But the rasterplots I uploaded before are the results of well 0000. If you download the wells.csv file indicating the wells and recordings (also in the gmail drive), the pipeline will run automatically across all the wells.
I think the pattern of synchronized events in the raster plot after sorting looks very similar to the electrode-level data, which is great. However, it seems that some electrode-synchronized events are not detected at the neuronal level. I’m not sure if this is an issue or not; perhaps the sorted data isn't expected to show exactly the same activity pattern.
I’m still unsure about the relationship between both levels of data (electrode and neuronal-sorted), and this might be something not fully understood yet. However, I would like to  check if I can trust the sorting results to possibly establish the connection.
Thank you very much for your help

yger, 2024-09-17:
There are some issues in your script, that could be simplified. I guess it used to be also valid for circus1, but with circus2 things are simpler. The raster I have after the run of circus2 (and no curation looks like that). Is this what do you expect?
[image]

sofiapuvogelvittini, 2024-09-17:
I get a very similar raster plot without curation, so it's good to know that we're getting the same output. However, in both your sorted output and mine, I can visually count 22 synchronized events (involving all neurons), whereas in the electrode-level data, I count 35 synchronized events in total. As I mentioned earlier, I'm still unsure if I should expect the same number of synchronized events after sorting. Maybe I should not expect same behaviour and there is no issue at all. 

In your paper of spuking circus 1, I  noticed that in the clustering analysis per electrode, there's a maximum of 10 clusters (assuming a maximum of 10 neurons surrounding each electrode). Do you think increasing this limit might help identify more synchronized events, or could it lead to the detection of non-real neurons?

Lastly, could you please point out the issues you found with my script? or let me know where I can find a more updated tutorial on SC2? I tried to follow the online tutorials to make the script.

Thanks a lot Pierre for your time and help!

sofiapuvogelvittini, 2024-09-30:
Hi @yger , I was wondering if there is any tutorial for SC2, so I can resolve the issues you mentioned
Thanks a lot!