SpikeInterface/spikeinterface #3400: Question Regarding Merging Units

JacobKelley101, 2024-09-12:
Hello, apologies if this question is naive, but I am having a hard time understanding the purpose of the merging functions available for the spike interface analyzer objects. I am running mountainsort5 and functionally things are working well, but sometimes it will over-sort some units. Minutes 0-20 (out of 100) of my recording will be labeled as one cell, then 20-60 another, then 60-100 a third. This isn't a big deal, the raster plots, waveforms, and locations make it very obvious that these cells need to be merged. The only way to merge units, though, is with the Analyzer, which (as far as I can tell) does not store spike time information. So even if one merges cells in the analyzer, the sorting object output by mountainsort5 still has the spike times listed in the original separate units. It's possible I am just too new to spike sorting, but if I want to combine the spike times to match the merged unit stored in the analyzer, is that something I just have to do myself? Or is there a function that will adjust the sorting object to match the data represented in a new, merged-analyzer?

alejoe91, 2024-09-12:
The analyzer stores spike times ;) The new `sorting_analyzer.merge_units()` function will do the merging for you so that the spikes of merged clusters are also combined

chrishalcrow, 2024-09-12:
You can get the spikes using `sorting_analyzer.to_spike_vector()`. Note that you can find all the method which apply to your `sorting_analyzer` by running  `dir(sorting_anayzer)` - there's a lot!

JacobKelley101, 2024-09-12:
@chrishalcrow when I try to call the analyzer.to_spike_vector() method It just says it doesn't exist. I printed the available methods with dir(analyzer) and don't see it in there either. I attatched the output of the dir(analyzer) command. Any ideas why I can't find my spike times? I tried analyzer.sorting, but all it shows are the unit IDs and the group labels, no other information that I can discern. 

['__class__',
 '__delattr__',
 '__dict__',
 '__dir__',
 '__doc__',
 '__eq__',
 '__format__',
 '__ge__',
 '__getattribute__',
 '__gt__',
 '__hash__',
 '__init__',
 '__init_subclass__',
 '__le__',
 '__lt__',
 '__module__',
 '__ne__',
 '__new__',
 '__reduce__',
 '__reduce_ex__',
 '__repr__',
 '__setattr__',
 '__sizeof__',
 '__str__',
 '__subclasshook__',
 '__weakref__',
 '_get_zarr_root',
 '_recording',
 '_save_or_select_or_merge',
 '_temporary_recording',
 'channel_ids',
 'channel_ids_to_indices',
 'compute',
 'compute_one_extension',
 'compute_several_extensions',
 'copy',
 'create',
 'create_binary_folder',
 'create_memory',
 'create_zarr',
 'delete_extension',
 'extensions',
 'format',
 'get_channel_locations',
 'get_computable_extensions',
 'get_default_extension_params',
 'get_dtype',
 'get_extension',
 'get_loaded_extension_names',
 'get_num_channels',
 'get_num_samples',
 'get_num_segments',
 'get_num_units',
 'get_probe',
 'get_probegroup',
 'get_recording_property',
 'get_saved_extension_names',
 'get_sorting_property',
 'get_sorting_provenance',
 'get_total_duration',
 'get_total_samples',
 'has_extension',
 'has_recording',
 'has_temporary_recording',
 'is_read_only',
 'is_sparse',
 'load',
 'load_all_saved_extension',
 'load_extension',
 'load_from_binary_folder',
 'load_from_zarr',
 'merge_units',
 'rec_attributes',
 'recording',
 'remove_units',
 'return_scaled',
 'sampling_frequency',
 'save_as',
 'select_units',
 'set_temporary_recording',
 'sorting',
 'sparsity',
 'unit_ids']

alejoe91, 2024-09-12:
> @chrishalcrow when I try to call the analyzer.to_spike_vector() method It just says it doesn't exist. I printed the available methods with dir(analyzer) and don't see it in there either. I attatched [...]

Right! And sorry @chrishalcrow were missing a subfield! It's `sorting_analyzer.sorting.to_spike_vector()`

JacobKelley101, 2024-09-12:
Ok that makes sense. I see now that the entire sorting object is now stored in the analyzer. So any functions that can be called on using the sorting object, (such as sorting.to_spike_vector or sorting.get_all_spike_trains) can also be called by adding the analyzer in front (analyzer.sorting.whatever_function). Thank you for helping me figure this out. Where in the "Read the Docs" website does it say that the analyzer stores all the same functions as the sorter? I read through the document a few times and couldn't figure it out, hence my post.

zm711, 2024-09-12:
I don't think it would accurate to say it has all the same functions as the `sorting`. Instead it stores a copy of your `sorting`, which can be modified as you perform operations. For example if you merge units you change the values in the `sorting`. Then if you needed to use a standard `sorting` function like `to_spike_vector` you could just call it on the sorter inside the analyzer. 

I think you raise a good point that we probably haven't documented merging well enough on RTD since it was a relatively recent addition to the analyzer. I can work on that soon!

JacobKelley101, 2024-09-12:
OK, thanks for clearing that up. If I were to perform merging on units I understand that the spike times in the internal sorter of the analyzer would be updated, but not the original sorter used to make the analyzer. Will there be a history stored somewhere that defines precisely which units taken from the original sorter (output by mountainsort) have been merged to create the new unit in the internally stored sorter of the analyzer?

zm711, 2024-09-12:
That's a great question. I think @alejoe91 will have to answer that one. If you use sortingview or spikeinterface-gui you'll get a json file that indicates all the merges. If you use our `get_potential_auto_merge` you could save those yourself as csv or json. And if you're doing your own merges then you could also save them yourself. Are you worried about needing to revert? Or just wanting a history?

samuelgarcia, 2024-09-13:
Yep. The merging (auto and manual) is pretty a new born. the effort on doc and features need to continued.

JacobKelley101, 2024-09-16:
@zm711 yeah I am interested in having a written record of which cells were merged, just in case. Great, well I think that answers all of my questions on this topic for now. I am going to close this issue, but if I encounter any related issues I will be sure to reference this tread. Thank you!

closed 2024-09-16 by JacobKelley101