SpikeInterface/spikeinterface #3410: correct_motion can't be set False in sorting =

BEGINRX, 2024-09-14:
I have a problem. The error information is following:
```
Traceback (most recent call last):
  File "/home/NITD/lxr/code/spike_sorting/run_sorter0913.py", line 628, in <module>
    start_sorting(manual,
  File "/home/NITD/lxr/code/spike_sorting/run_sorter0913.py", line 530, in start_sorting
    sorting = run_sorter_sc2(rec_w_sub, detection, general, selection, matching, apply_preprocessing, output_folder=sorter_path)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/lxr/code/spike_sorting/run_sorter0913.py", line 347, in run_sorter_sc2
    sort_res = si.run_sorter('spykingcircus2', recording, 
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py", line 216, in run_sorter
    return run_sorter_local(**common_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py", line 276, in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 301, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/internal/spyking_circus2.py", line 116, in _run_from_folder
    recording = cls.load_recording_from_folder(sorter_output_folder.parent, with_warnings=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 212, in load_recording_from_folder
    recording = load_extractor(json_file, base_folder=output_folder)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1185, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 769, in load
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 515, in from_dict
    extractor = _load_extractor_from_dict(dictionary)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1109, in _load_extractor_from_dict
    new_kwargs[name] = [transform_dict_to_extractor(e) for e in value]
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1109, in <listcomp>
    new_kwargs[name] = [transform_dict_to_extractor(e) for e in value]
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1101, in <lambda>
    transform_dict_to_extractor = lambda x: _load_extractor_from_dict(x) if is_dict_extractor(x) else x
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1105, in _load_extractor_from_dict
    new_kwargs[name] = _load_extractor_from_dict(value)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1124, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sortingcomponents/motion/motion_interpolation.py", line 302, in __init__
    assert channel_locations.ndim >= motion.dim, (
                                     ^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'dim'
```
Then I set `apply_motion_correction=False`, like 
```
sort_res = si.run_sorter('spykingcircus2', recording, 
                             detection=detection, 
                             general=general,
                             selection=selection,
                             matching=matching,
                             apply_preprocessing=apply_preprocessing,
                             apply_motion_correction=False,
                             remove_existing_folder=True, 
                             verbose=False, 
                             output_folder=output_folder)
```
But it doesn't seem to work.

zm711, 2024-09-14:
We've had this issue before but I keep forgetting what is causing it. @samuelgarcia can comment on the issue. It might be fixed on main if your willing to install from source. Is that a possibility for you?

alejoe91, 2024-09-16:
@BEGINRX we fixed this in main, see #3313 and #3341

We are going to make a release today including this bug fix. We'll ping you when it's released so you can upgrade and try again

zm711, 2025-02-05:
@alejoe91 was this in last release or will this be in 0.102.0?