SpikeInterface/spikeinterface #3413: Error running spykingcircus, due to space in recording file path

Djoels, 2024-09-15:
Running spykingcircus in docker mode (probably in regular mode as well) results in an error if there is a space in the path.

This probably relates to the shell command not putting the path in quotes:

https://github.com/SpikeInterface/spikeinterface/blob/082bb399cd9e7fba3364c358500da8af2b119c69/src/spikeinterface/sorters/external/spyking_circus.py#L157-L169

Is it mandatory that the path has no spaces? I can not really enforce this from an automation perspective in my own experiment pipeline...

If it's acceptable to put path in quotes (and this wouldn't break anything else) then I'm willing to make a PR for this.

alejoe91, 2024-09-15:
Generally speaking, spaces in paths are a nightmare to deal with.. I would try to enforce it ad much as I could, but feel free to also make a PR with added quotes. I think that being extra cautious avuto this doesn't hurt!

Djoels, 2024-09-15:
In trying to fix this, I'm working from a local version to run spykingcircus in docker, using `installation_mode="Folder"` and setting the path using the `spikeinterface_folder_source` parameter. Does this path have to point to the root of the spikeinterface github repo, or the src/spikeinterface folder? Tried a couple of options already, error keeps popping up:

```
Starting container
Running spykingcircus sorter inside spikeinterface/spyking-circus-base
Stopping container
2024-09-15 20:12 - root         - ERROR    - Error occured during sorting: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/verpla38-gpu-v100/code/Users/verpla38/data/hybrid-gt/2023-0502_Ctl653xrep/Sparrow User-P220081-D11-J3-12_09_2023-11_09_17.sparrowdata/hybridgt_20240816/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'
Traceback (most recent call last):
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/verpla38-gpu-v100/code/Users/verpla38/ai-boc/src/hpoptuna.py", line 340, in main
    scores = _run_sorter(
             ^^^^^^^^^^^^
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/verpla38-gpu-v100/code/Users/verpla38/ai-boc/src/hpoptuna.py", line 138, in _run_sorter
    sorting = ss.run_sorter(sorter_name=sorter,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/anaconda/envs/azureml_py312/lib/python3.12/site-packages/spikeinterface/sorters/runsorter.py", line 210, in run_sorter
    return run_sorter_container(
           ^^^^^^^^^^^^^^^^^^^^^
  File "/anaconda/envs/azureml_py312/lib/python3.12/site-packages/spikeinterface/sorters/runsorter.py", line 663, in run_sorter_container
    raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/verpla38-gpu-v100/code/Users/verpla38/data/hybrid-gt/2023-0502_Ctl653xrep/Sparrow User-P220081-D11-J3-12_09_2023-11_09_17.sparrowdata/hybridgt_20240816/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'
```

zm711, 2024-12-03:
I'll ping @alejoe91 since he knows containers way better than I do :)

Djoels, 2024-12-04:
Hello, I've since moved to spykingcircus2, which gives really nice results. If you wish I can close this. Not sure if it still is problematic with latest releases... If need be I keep this open, but it's no longer related to the original request then.

zm711, 2024-12-04:
Okay I think we can close this one for now. As Alessio said it is a pain to deal with paths with spaces in general. If someone decides they want to try a PR for this they are welcome to!

closed 2024-12-04 by zm711