SpikeInterface/spikeinterface #3423: AttributeError: module 'spikeinterface.extractors' has no attribute 'RecordingExtractor'

Chiara-toschi, 2024-09-18:
Apologies if this is super naive, I'm kind of new to python, I'm using someone else's code to do some analyses. 

I run

`import spikeinterface.extractors as se`

and then further down in the code:

`def get_amp_files(self) -> (se.RecordingExtractor, pd.Series):`

and get this error:

```
in AmplifierChannels
    def get_amp_files(self) -> (se.RecordingExtractor, pd.Series):
AttributeError: module 'spikeinterface.extractors' has no attribute 'RecordingExtractor'
```

I'm running Version: 0.95.1 of spikeinterface and cannot update it because I think everything else will break

zm711, 2024-09-18:
That's pretty old. Sorry you can't update. @alejoe91 or @samuelgarcia would have to comment on the older side of the api. The best advice would honestly be to update in general. If you're new to python maybe this could be a learning experience to update the code while learning :) We are pretty responsive to questions as they come up!

chrishalcrow, 2024-09-18:
If you do think of updating your version, we have some good beginner resources. You could try working through the quickstart tutorial (https://spikeinterface.readthedocs.io/en/latest/get_started/quickstart.html) or download this Jupyter notebook from a recently conference in Edinburgh (https://github.com/SpikeInterface/SpikeInterface-Training-Edinburgh-May24/blob/main/overview_tutorials/main_tutorial.ipynb), which covers similar material.

zm711, 2024-12-03:
Since this is due to needing to using old versions of the code we will close this but as Chris wrote we have tons of help trying to update your code and you can feel free to open issues if they come up.

closed 2024-12-03 by zm711