SpikeInterface/spikeinterface #3425: How to get sync timestamps from spikeglx and open ephys

gabrielstine, 2024-09-18:
Hi Guys,

I have a simple question that I'm having trouble finding the answer to in the documentation. I am recording simultaneously from a neuropixels probe (recorded in spikeGLX) and a DBC probe (recorded in open ephys). I'd like to extract the timestamps of the sync events from each stream and save them as .npy files. In open ephys, the sync signal is a ttl pulse fed into channel 1. I imagine there are straightforward functions for extracting and saving these, but I can't find them!

Thanks!
Gabe

zm711, 2024-09-18:
I feel like @alejoe91 is going to need to be the person to comment on this. I don't actually use ephys or any of the event based code in spikeinterface. So just ping this issue if you don't get a response soon!

alejoe91, 2024-09-19:
Hi @gabrielstine !

You should be able to read the open ephys events with this function:
```
events = se.read_openephys_events(openephys_folder)
```

This will return an [Event](https://spikeinterface.readthedocs.io/en/latest/modules/core.html#event) object, which will allow you to interact with the event data. There can be events from multiple channel ids, which you can retrieve with the `events.channel_ids` field.
In particular, calling `evts = events.get_events(channel_id)` will return a structured numpy array with the following field:
`dtype([('time', '<f8'), ('duration', '<f8'), ('label', '<U100')])`. For open ephys, the `label` is the channel the event was recorded on, so you can select all events from channel 1 as follows:
```
evts_1 = evts[evts["label"] == "1"]
```

Let me know if you have other questions!

gabrielstine, 2024-09-19:
Thanks @alejoe91. And is there a way to automatically get the timestamps at which the sync channel went high in the spikeGLX data, or do I need to extract this manually? If not, could be nice to have a built-in function that takes the channel as input and returns timestamps.

alejoe91, 2024-09-19:
You can try the `event_spikeglx = se.read_spikeglx_events(spikeglx_folder)` ;)

chongtianyifa, 2024-10-12:
@alejoe91  It seems that se does not have read_spikeglx_event module.
[image]

alejoe91, 2024-10-12:
Did you try this?

```
import spikeinterface.extractors as se

event = se.read_spikeglx_event(path-to-sglx-folder)
```

It's a function and not a module. Can you paste the error you're getting?

chongtianyifa, 2024-10-13:
with se.read_spikeglx_event(path-to-sglx-folder), no error came out. However, no events were detected. My event signals were actually synchronization signal with 1s on and 29 s off.
[image]

alejoe91, 2024-10-13:
Maybe then you used the SYNC channel (i.e. the 385th signal)?

chongtianyifa, 2024-10-13:
Yes, the event was recorded in the SYNC channel. Is there a way to extract events from this channel?

gabrielstine, 2024-10-23:
@alejoe91 I'm still struggling with this, and not seeing anything in the docs. Feels like it should be very easy/simple given how ubiquitous this step is in neuropixels pipelines. To reiterate/clarify, channel 385 receives TTL pulses that are used to synchronize my spikeGLX data to other data streams. I need to extract this signal and find the timepoints at which it went "high." Previously I was using catGT for preprocessing, which does this automatically. Does something like the below exist?

```
rec_imec0 = read_spikeglx(spkGLX_folder, stream_name='imec0.ap')
sync_times_imec0 = get_sync_times_spikeglx(rec_imec0)
```

If not, could you walk me through how I would, at the very least, extract the signal on channel 385? I believe this is what @chongtianyifa is asking for as well.

alejoe91, 2024-10-24:
I see, sorry I misunderstood. @samuelgarcia was planning to work on such automated sync function for spikeGLX, so let's ping him!

You can extract the sync channel recording as follows:
```
recording_ap_with_sync = se.read_spikeglx(spkGLX_folder, stream_name='imec0.ap', load_sync_channel=True)
# only select the last SYNC channel
recording_sync = recording_ap_with_sync.select_channels([recording_ap_with_sync.channel_ids[-1]])
```

Then you can get the sync channel traces and look for rising edges:
```
traces_sync = recording_sync.get_traces()[:, 0] # the slicing is to get a 1d array

rising_edges = np.where(np.diff(traces_sync) > 0)[0]
```

gabrielstine, 2024-10-24:
Thanks Alessio! Appreciate all of the work you guys do.