SpikeInterface/spikeinterface #3442: Centralise the binary file saving metadata

JoeZiminski, 2024-09-25:
Following on from a discussion at #3245. At present there is a `binary.json` metadata file written during the saving of binary data. This duplicates the `si_folder.json` metadata which has caused some problems with failure to propagate metadata and confusing behaviour during debugging #3245. The logic flow for the data-saving is discussed [here](https://github.com/SpikeInterface/spikeinterface/pull/3245#issuecomment-2341614288). It would be nice to completely deprecate `binary.json` and use only `si_folder.json` to centralise the metadata.

jmamodeo, 2026-11-06:
I’d be happy to help with this. Since [#4671](https://github.com/SpikeInterface/spikeinterface/pull/4671) moved properties, annotations, and probe metadata into separate files, is deprecating `binary.json` still the intended direction?

Would you prefer keeping it as the binary-storage configuration, with `si.load(folder)` documented as the way to restore the complete recording, or migrating away from it while retaining support for older folders?

I’d suggest a small first PR for the agreed direction, with regression tests covering properties and annotations on save/load, plus legacy-folder compatibility. I’d leave new metadata fields and fingerprints out of scope. Happy to put together the implementation once the direction is clear.

JoeZiminski, 2026-11-11:
Hi @jmamodeo thanks for this. This is quite old now, I am not certain if the discussion still applies, although it would be nice to avoid the duplication if possible. Can you create a report determining if the problem is still valid, and what a solution would look like? Ideally, if there is a neat solution that does not touch too much of the codebase, we could continue with this idea. thanks!