SpikeInterface/spikeinterface #3444: Containers do not appear to include Wine64, zugbruecke for PL2 files

nikhilchandra, 2024-09-25:
I might not fully understand how everything works, but would a Docker container (and presumably a Singularity container) for one of the spike sorters (say, spikeinterface/kilosort4-base) need to include Wine64 and zugbruecke in order to work with PL2 files? Out of curiosity I tried running the run_sorter method for kilosort4 with docker_image=True and got an error message that indicated the absence of Wine.

zm711, 2024-09-26:
yes it would. the docker containers (singularity I don't know) are going to be linux-esque. So if you wanted to use docker for plexon2 stuff you would need to include wine in the extras. @alejoe91 knows way more about the docker stuff than me so he could give you specifics for commands.

nikhilchandra, 2024-09-30:
@alejoe91 Is it unfeasible to include wine64 and pickle in the Docker container from the get-go? This would the negate the need for installing on the fly and also avoid conversion to BIN.

alejoe91, 2024-09-30:
The containers tend to have minimal installations of the sorters dependencies only, so that they are easier to maintain and upgrade.

What you could do is build a docker image `FROM` a spikeinterface container and add the additional wine dependencies and distribute tham as something like: `spikeinterface-pl2/kilosort4-base`, etc.

When running a sorter, then you can specify the exact image to use as a container!

zm711, 2024-12-03:
Closing since I think this has been answered. Reopen if needed!

closed 2024-12-03 by zm711