SpikeInterface/spikeinterface #3450: Error at "random_spikes_selection"

borrepp, 2024-09-29:
Hello, I think I found an error at "core.sorting_tools.random_spikes_selection". 

A note about how I came to this error:
I was trying to create a sorting_analyzer from a single electrode recording with only one cluster and very few spikes (n=10). This was because of a high 'detect_threshold' value. When I changed the value and more spikes were detected, everything worked fine. However, I found the following mismatch when selecting random spikes:

At the step when the spike vector is converted to indices (core.sorting_tools ~ line: 200):

`spike_indices = spike_vector_to_indices(spikes, sorting.unit_ids) `

I got the following:
`
spike_indices =
{0: {1: array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9], dtype=int64)}}`

Which makes sense given that I only have 10 spikes. 

However, when removing the spikes from the margin, the code is still using sample_index, not relative indices. In other words, the variable "margin_size" (which removes spikes from each border of segments) is defined in units of the number of samples to avoid the edges of the segment, but the code uses relative indexes to remove it (line 206):

```
inds_in_seg = spike_indices[segment_index][unit_id] + cum_sizes[segment_index]
if margin_size is not None:
    inds_in_seg = inds_in_seg[inds_in_seg >= margin_size]
    inds_in_seg = inds_in_seg[inds_in_seg < (num_samples[segment_index] - margin_size)] 
```
 
In my example data of 10 spikes, I got an empty array from "random_spikes_selection" because my "margin_size" is 18 (ms_after = 1.6 miliseconds at 30KHz) I don't have any index higher than 18 (my indexes are 0 to 9). But it shouldn't be based on the relative index, it should be on the sample index of the spikes.

 I think this part of the code is a legacy of using "spike_vector_to_spike_trains" instead of "spike_vector_to_indices". Because, when when I use:
`
spike_samples = spike_vector_to_spike_train(spikes, sorting.unit_ids) `

I got:
```
spike_samples = 
{0: {1: array([   251331,   3037977,   3237922,  13033056,  21827320,  24649558,
        35249946,  53788634,  70662738, 103737779], dtype=int64)}}
```

Which makes sense for the computations related to margin_size. I think the computation of removing spike indexes from the margin, needs to be updated. Otherwise, it will always remove the first "margin_size" spikes, and it will never remove spikes close to the end of the segment. 

Hope it makes sense these comments. 

Thanks. 
Pepe

borrepp, 2024-09-30:
Hello @samuelgarcia 

I added some changes to the code (not committing any changes, just local on my computer) trying to "fix" the problem. I would like to get feedback on whether this makes sense while I wait to update the next release. 

At line 200, after calling the function "spike_vector_to_indices" I added this:

```
# Extract spike_trains to get sample_index
spike_trains = spike_vector_to_spike_trains(spikes, sorting.unit_ids)
```

Then, to remove spikes at the margins of the segment I used the spike trains:

```
if margin_size is not None:
    inds_in_seg = inds_in_seg[spike_trains[segment_index][unit_id] >= margin_size]
    inds_in_seg = inds_in_seg[spike_trains[segment_index][unit_id] < (num_samples[segment_index] - margin_size)]
```

Not sure if computing "spike_vector_to_spike_trains" will add too much computation time.

Thanks

samuelgarcia, 2024-10-01:
Hi José.
thanks a lot for catching this. This was a big mistake form me.
Here a fix #3456. COudl you test it ?

borrepp, 2024-10-01:
Hi Samuel, 

It worked fine¡¡¡

Thank you so much.

PR #3456 (2024-10-02): fix random_spikes_selection()
fix random_spikes_selection()
fix #3450

@borrepp could you check this please ?
closed 2024-10-02 by alejoe91