SpikeInterface/spikeinterface #3451: Need to save analyzer extension's run_info before run starts (bug from #3347?)

jonahpearl, 2024-09-29:
> @ alejoe91, can you explain why this call to `self._save_run_info()` line got deleted? I think it's important to have it saved before the run starts, otherwise if the run crashes there is no run_info present to be loaded. 

 _Originally posted by @jonahpearl in [ff07ac6](https://github.com/SpikeInterface/spikeinterface/commit/ff07ac603bb8552ef40bd1f43962f201a72df17c#r147358628)_

jonahpearl, 2024-09-29:
Sorry idk how well the commenting works, but here's a screenshot of the line i'm referencing, specifically the deletion of line 2017:
[image]

alejoe91, 2024-09-30:
The run info dict is instantiated with the object, so there is no need to save it before it has run.

Is it triggering a specific error for you? All tests are running fine, so I don't think it's crashing

jonahpearl, 2024-10-08:
It doesn't accomplish the intended use case, which was to allow the user (or the code) to check the run info dict for run_completed as being true or false in case the analyzer code crashes. As it is, one could check for the existence of the run info json file as a proxy for whether the run finished, I suppose, but that's a bit counterintuitive.