SpikeInterface/spikeinterface #3453: offline sorting in a container

rat-h, 2024-09-30:
If I try to run sorting on a computer that isn't connected to the internet (offline) and this sorting is installed in the environment (such as kilosort 4, tridesclous 2, spykingcircus2), my scripts work flawlessly. 

However, any attempt to run a sorter from container (kilosort 2.5, hdsor, etc.) ends with an error "spikeinterface not found". I think, it is because, at every container initiation, scripts install spikeinterface from pypi. So, if the computer is offline, it cannot use containerized sorters. If this is correct, can I give access to my local pip cache and avoid calling pypi?

We run singularity containers, a requirement for any HPS cluster nowadays.

samuelgarcia, 2024-10-01:
Hi.
Yes spikeinterface is installed on the fly in the container.
So if you do not have internet connection on run time you have two choices:
  1. build a contanier on top of the one we distribute with the desired spikeinterface version inside so the installaiton is 
      skiped
  2. Play with the     `installation_mode="folder"` and `spikeinterface_folder_source=None` options in 
      `run_sorter_container()`. So you can force installation from an existing local folder which is the local copy of 
       spikeinterface repo.

rat-h, 2024-10-01:
Thank you, @samuelgarcia.
Just a suggestion, if spikeinterface can reuse a container instead of spin a new one from an image, this problem could be easily solved.

closed 2024-10-01 by rat-h

alejoe91, 2024-10-02:
> Thank you, @samuelgarcia. Just a suggestion, if spikeinterface can reuse a container instead of spin a new one from an image, this problem could be easily solved.

@rat-h could you make a PR? I agree this could be a viable solution!