SpikeInterface/spikeinterface #3454: Can I concate two recordings from one probe and perform motion correction to restore them to one recording?

DaohanZhang, 2024-10-01:
Hey!  An system error occurred during when I was using neuropixels for recordings. I restarted the system and got two recordings from one probe. There will be some drifting due to the interval from the end of my first recordings to the beginning of my second reording, which lasts for approximately 10 min. I wonder is it possible to concate two recordings together and use motion correction, in order to restore them as one recording for later sorting process? Thanks a lot!

chrishalcrow, 2024-10-02:
Hi @DaohanZhang , this problem is being worked on (mainly by @JoeZiminski) here: https://github.com/SpikeInterface/spikeinterface/pull/3126 but this is still in development.

Until that is ready, you could just try and concatenate the recordings using `both_recordings = concatenate_recordings([recording_1, recording_2])` (see more [here](https://spikeinterface.readthedocs.io/en/latest/tutorials/core/plot_5_append_concatenate_segments.html)), apply motion correction (or nothing) to `both_recordings` and see what happens.

cwindolf, 2024-10-16:
Hi @DaohanZhang , I'd second what Chris has written there! This is a strategy I've used for some multi-segment recordings in the past.

DaohanZhang, 2024-10-19:
Thanks @cwindolf! So you got consistent number of units in each segments after sorting process,  just like sorting one segments, right?

closed 2024-11-03 by DaohanZhang