SpikeInterface/spikeinterface #3458: Motion correction error: win_step_um=400.0/win_margin_um=-200.0 are too large for the probe size

desaponte, 2024-10-02:
Hello!
I'm following the spike interface tutorial on testing different motion correctors on my NP2 4 shank recording.  

The kilosort-like correction works..... but when I run dredge , I get a probe size issue.

desaponte, 2024-10-02:
Sure!  Please see below, thanks for the help 

---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[17], line 15
     13 if folder.exists():
     14     shutil.rmtree(folder)
---> 15 recording_corrected, motion, motion_info = si.correct_motion(
     16     rec, preset=some_presets, folder=folder, output_motion=True,output_motion_info=True,**job_kwargs)

File ~/miniconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/preprocessing/motion.py:436, in correct_motion(recording, preset, folder, output_motion, output_motion_info, overwrite, detect_kwargs, select_kwargs, localize_peaks_kwargs, estimate_motion_kwargs, interpolate_motion_kwargs, **job_kwargs)
    429     run_times = dict(
    430         detect_peaks=t1 - t0,
    431         select_peaks=t2 - t1,
    432         localize_peaks=t3 - t2,
    433     )
    435 t0 = time.perf_counter()
--> 436 motion = estimate_motion(recording, peaks, peak_locations, progress_bar=progress_bar, **estimate_motion_kwargs)
    437 t1 = time.perf_counter()
    438 run_times["estimate_motion"] = t1 - t0

File ~/miniconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion/motion_estimation.py:111, in estimate_motion(recording, peaks, peak_locations, direction, rigid, win_shape, win_step_um, win_scale_um, win_margin_um, method, extra_outputs, progress_bar, verbose, margin_um, **method_kwargs)
    108     extra = None
    110 # run method
--> 111 motion = method_class.run(
    112     recording,
    113     peaks,
    114     peak_locations,
    115     direction,
    116     rigid,
    117     win_shape,
    118     win_step_um,
    119     win_scale_um,
    120     win_margin_um,
    121     verbose,
    122     progress_bar,
    123     extra,
    124     **method_kwargs,
    125 )
    127 if extra_outputs:
    128     return motion, extra

File ~/miniconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion/dredge.py:116, in DredgeApRegistration.run(cls, recording, peaks, peak_locations, direction, rigid, win_shape, win_step_um, win_scale_um, win_margin_um, verbose, progress_bar, extra, **method_kwargs)
    113 except ImportError:
    114     raise ImportError("The dredge method require torch: pip install torch")
--> 116 outs = dredge_ap(
    117     recording,
    118     peaks,
    119     peak_locations,
    120     direction=direction,
    121     rigid=rigid,
    122     win_shape=win_shape,
    123     win_step_um=win_step_um,
    124     win_scale_um=win_scale_um,
    125     win_margin_um=win_margin_um,
    126     extra_outputs=(extra is not None),
    127     progress_bar=progress_bar,
    128     **method_kwargs,
    129 )
    131 if extra is not None:
    132     motion, extra_ = outs

File ~/miniconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion/dredge.py:310, in dredge_ap(recording, peaks, peak_locations, direction, rigid, win_shape, win_step_um, win_scale_um, win_margin_um, bin_um, bin_s, max_disp_um, time_horizon_s, mincorr, do_window_weights, weights_threshold_low, weights_threshold_high, mincorr_percentile, mincorr_percentile_nneighbs, amp_scale_fn, post_transform, histogram_depth_smooth_um, histogram_time_smooth_s, avg_in_bin, thomas_kw, xcorr_kw, device, progress_bar, extra_outputs, precomputed_D_C_maxdisp)
    307 contact_depths = recording.get_channel_locations()[:, dim]
    308 spatial_bin_centers = 0.5 * (spatial_bin_edges_um[1:] + spatial_bin_edges_um[:-1])
--> 310 windows, window_centers = get_spatial_windows(
    311     contact_depths,
    312     spatial_bin_centers,
    313     rigid=rigid,
    314     win_shape=win_shape,
    315     win_step_um=win_step_um,
    316     win_scale_um=win_scale_um,
    317     win_margin_um=win_margin_um,
    318     zero_threshold=1e-5,
    319 )
    321 # TODO charlie : the count has disapeared
    322 # if extra_outputs and count_masked_correlation:
    323 #     extra["counts"] = counts
    324 
    325 # cross-correlate to get D and C
    326 if precomputed_D_C_maxdisp is None:

File ~/miniconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion/motion_utils.py:312, in get_spatial_windows(contact_depths, spatial_bin_centers, rigid, win_shape, win_step_um, win_scale_um, win_margin_um, zero_threshold)
    309 num_windows = int((max_ - min_) // win_step_um)
    311 if num_windows < 1:
--> 312     raise Exception(
    313         f"get_spatial_windows(): {win_step_um=}/{win_scale_um=}/{win_margin_um=} are too large for the "
    314         f"probe size (depth range={np.ptp(contact_depths)}). You can try to reduce them or use rigid motion."
    315     )
    316 border = ((max_ - min_) % win_step_um) / 2
    317 window_centers = np.arange(num_windows + 1) * win_step_um + min_ + border

Exception: get_spatial_windows(): win_step_um=400.0/win_scale_um=400.0/win_margin_um=-200.0 are too large for the probe size (depth range=705.0). You can try to reduce them or use rigid motion.

samuelgarcia, 2024-10-04:
For small probe you should try rigid motion correction, if you think you really need non rigid then you should lower `win_step_um` and `win_scale_um` `win_margin_um`. Theses 3 params define the number of gaussian window for non rigid motion correction on  depth axis.

cwindolf, 2024-10-16:
Hi @desaponte for NP2.4, I'd recommend a couple of things. First off, I've had the best results when doing motion correction on each shank separately -- you may already be doing that, not sure. Second, there are parameters recommended for that probe in the DREDge paper which you can check out -- that would just mean setting `win_step_um=100` and `win_margin_um=-150`. Sometimes trying `250` for the scale can help too. Feel free to send me an email if you want some more detailed advice on parameters! (email on my github profile)

BEGINRX, 2024-11-26:
I want to know these parameters' meaning. Do you have a picture to illustrate the locations on the probe?

BEGINRX, 2024-12-03:
> Hi @BEGINRX , I don't have a picture, but there is a section discussing what these paramters are in the preprint (https://www.biorxiv.org/content/10.1101/2023.10.24.563768v1) starting halfway down page 49

Thank you!