SpikeInterface/spikeinterface #3459: error when using .merge(pairs)

sofiapuvogelvittini, 2024-10-03:
Hi, I have been  using the function scur.get_potential_auto_merge() to get pairs f neurons to be merged and then I merge them with .merge(pairs). It works well most of the time, but now  I am getting this error "AssertionError: new_unit_ids should have the same len as merge_unit_groups. The output of pairs is [(520, 1001, (861,872)]. I can identify this units in the sorting object.  Attahced the full error in a pdf.
Do you have any thoughts on what could be going wrong this time?

Thanks a lot!
[merge_error.pdf](https://github.com/user-attachments/files/17241188/merge_error.pdf)

sofiapuvogelvittini, 2024-10-03:
I solved it manually by merging each unit and assigning a new ID, like this
curation_sorting.merge([(520)],1008)
curation_sorting.merge([(861)],1009)
Could it be a bug in the merge function when handling multiple units?
Best!

zm711, 2024-10-03:
I haven't seen the nested merge pairs before. Could you post a bit more of your script? And which version of spikeinterface you're using?

sofiapuvogelvittini, 2024-12-31:
I am very sorry for the delay, I missed these emails. 
Here is how I made it work with my data:

keep_mask= (qm_data["snr"]>snr_th)#(qm_data["sd_ratio"]<sd_ratio) & (qm_data["snr"]>snr_th)
full_path_curated=os.path.join(output_folder,"analyzer_SC2/waveforms_curated")
        
  if keep_mask.any():
                    
                     curated_analyzer=analyzer_SC2.select_units(analyzer_SC2.unit_ids[keep_mask])
                     curated_sorting=sorting_SC2.select_units(sorting_SC2.unit_ids[keep_mask])
                     pairs= scur.get_potential_auto_merge(curated_analyzer)
                     curation_sorting=scur.CurationSorting(curated_sorting)
                     n_pairs=len(pairs)

                         if n_pairs == 1:
                         print("merging pairs")   
                         curation_sorting.merge(pairs)
                         curation_sorting=curation_sorting.sorting
                                         
                    
                    
                     **elif n_pairs> 1:**
                         print("more than one pair to merge")
                         last_unit=curated_sorting.get_unit_ids()[-1]
                         new_unit_first=last_unit+1
                         new_unit_last=last_unit+n_pairs
                         new_units=[]
                    
                         while new_unit_first <= new_unit_last:
                             new_units.append(new_unit_first)
                             new_unit_first+=1
                           
                         for p, new_unit in zip(pairs, new_units):
                             print("merging pair:",p, " in unit:",new_unit)
                             curation_sorting.merge(p, new_unit)
                         curation_sorting=curation_sorting.sorting
                                
                     else: 
                         print("no pairs to be merged")
                         curation_sorting=curation_sorting.sorting

samuelgarcia, 2025-01-06:
Hi.
Curation_sorting will depracted one day and totally unmaintened. You should not use it.

The correct is to use the `Analyzer.merge(merge_unit_groups)`.
By default the new units id are generated.

The output pairs you have `[(520, 1001, (861,872)]` is really strange it should be a list of list.
I think this is the problem you have.

Note that we did some recent update in get_potential auto merge().
This function will be depretcated.
Use the new function `compute_merge_unit_groups()` instead.

closed 2025-04-04 by zm711