SpikeInterface/spikeinterface #3468: Full test failing

zm711, 2024-10-08:
@samuelgarcia @yger
```
FAILED src/spikeinterface/benchmark/tests/test_benchmark_sorter.py::test_SorterStudy - AssertionError
```
Sam let us know about the anarchy. Just so you both keep the testing in mind :)

yger, 2024-10-08:
Arg damned. Anarchy is taking over, let's bring back order here @samuelgarcia this is related to your refactoring, I can try to have a look tomorrow.

samuelgarcia, 2024-10-09:
sorry for that.
I swear this was green when I merged it!

zm711, 2024-10-10:
Still failing on full tests. I don't know if this is a state issue with how our full tests are running, but same error is still present. I'll close once we see the tests pass. Maybe the commit hasn't made it into the full tests yet. If it is green today then we close otherwise this might need a bit of a deeper read. Let's see.

PR #3469 (2024-10-09): Fix main
fix #3468.
@zm711 : fortunatly anarchy rebelion are rapidely nipped in the bud.
closed 2024-10-10 by zm711