SpikeInterface/spikeinterface #3471: Error when loading template_metrics extension

jonpedros, 2024-10-10:
I've been getting some errors when computing/loading extensions from a `SortingAnalyzer` containg a `template_metrics` extension. I calculate extensions from a dict as follows:

```py
phy_exts = {"random_spikes": {},
            "waveforms": {},
            "templates":{},
            "noise_levels": {},
            "principal_components": {"n_components":5, "mode":"by_channel_local"},
            "template_similarity": {"method": "cosine_similarity"},
            "spike_amplitudes": {},
            "quality_metrics": {},
            "template_metrics": {}}

analyzer_obj.compute(phy_exts)
```

After successfully calculating waveforms, fitting PCAs and projecting waveforms, I get these errors:

```py
c:\Users\systemses\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\sortinganalyzer.py:1924: UserWarning: Found no run_info file for template_metrics, extension should be re-computed.
  warnings.warn(f"Found no run_info file for {self.extension_name}, extension should be re-computed.")
c:\Users\systemses\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\sortinganalyzer.py:1931: UserWarning: Found no run_info file for template_metrics, extension should be re-computed.
  warnings.warn(f"Found no run_info file for {self.extension_name}, extension should be re-computed.")
c:\Users\systemses\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\sortinganalyzer.py:2004: UserWarning: Found no data for template_metrics, extension should be re-computed.
  warnings.warn(f"Found no data for {self.extension_name}, extension should be re-computed.")
```

After which it will proceed to calculate `spike_amplitudes` successfully but then hang indefinitely.

chrishalcrow, 2024-10-10:
Hello,
Thanks for posting the issue. I can't reproduce the error. Here was my attempt to reproduce:

```
import spikeinterface.full as si

rec, sort = si.generate_ground_truth_recording()
analyzer_obj = si.create_sorting_analyzer(recording=rec, sorting=sort, format="binary_folder", folder="test_analyzer")

phy_exts = {"random_spikes": {},
            "waveforms": {},
            "templates":{},
            "noise_levels": {},
            "principal_components": {"n_components":5, "mode":"by_channel_local"},
            "template_similarity": {"method": "cosine_similarity"},
            "spike_amplitudes": {},
            "quality_metrics": {},
            "template_metrics": {}}

analyzer_obj.compute(phy_exts)

analyzer_2 = si.load_sorting_analyzer("test_analyzer")
analyzer_2.compute(phy_exts)
```

I get the warning, but the spike amplitudes are successfully computed, then it computes the pc_metrics. It'd be good if I could reproduce your problem. Do you know which extensions were successfully computed for the analyzer that you're loading?

I'll look into the warning, which seems like it shouldn't be there.

zm711, 2024-10-10:
Hey Chris are you on windows? Maybe I should test. There can some folder release issues with Windows that doesn't happen on Mac/Linux (which I think you mostly test from). I'm not free until tomorrow, but happy to also troubleshoot then unless you have a windows computer to work on :)

jonpedros, 2024-10-15:
If I disable computing of both template_metrics and quality_metrics, code runs fine. If I disable only template_metrics, code still hangs indefinitely but I get this error at some point: 

```py
[c:\Users\systemses\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\qualitymetrics\misc_metrics.py:908](file:///C:/Users/systemses/Anaconda3/envs/si_env/Lib/site-packages/spikeinterface/qualitymetrics/misc_metrics.py:908): UserWarning: Some units have too few spikes : amplitude_cutoff is set to NaN
  warnings.warn(f"Some units have too few spikes : amplitude_cutoff is set to NaN")
[c:\Users\systemses\Anaconda3\envs\si_env\Lib\site-packages\numpy\core\_methods.py:206](file:///C:/Users/systemses/Anaconda3/envs/si_env/Lib/site-packages/numpy/core/_methods.py:206): RuntimeWarning: Degrees of freedom <= 0 for slice
  ret = _var(a, axis=axis, dtype=dtype, out=out, ddof=ddof,
[c:\Users\systemses\Anaconda3\envs\si_env\Lib\site-packages\numpy\core\_methods.py:163](file:///C:/Users/systemses/Anaconda3/envs/si_env/Lib/site-packages/numpy/core/_methods.py:163): RuntimeWarning: invalid value encountered in divide
  arrmean = um.true_divide(arrmean, div, out=arrmean,
[c:\Users\systemses\Anaconda3\envs\si_env\Lib\site-packages\numpy\core\_methods.py:198](file:///C:/Users/systemses/Anaconda3/envs/si_env/Lib/site-packages/numpy/core/_methods.py:198): RuntimeWarning: invalid value encountered in divide
  ret = ret.dtype.type(ret / rcount)
```

chrishalcrow, 2024-10-15:
I managed to debug the errors, I think! This could be the cause of the hanging, or maybe not. I would appreciate it if you could try out the newly linked PR #3478 - thanks!

PR #3478 (2025-01-31): Only load `template_metrics` extension on compute if keeping some metrics
Might fix #3471
(Could you try this out please @jonpedros)

When `template_metrics` is computed and `delete_existing_metrics = False`, any old metrics which aren't being recomputed are kept. To do this, the `_run` method loads the old template metric extension (if it exists). However, if this is the first time it has run, it has already created the extension folder. In this case `_run` sees the newly-created folder and tries to load it. Bu there's not much in it, so a no `run_info` warning appears.

To avoid this, this P [... cut]
closed 2025-01-31 by alejoe91