SpikeInterface/spikeinterface #3484: Process a recording by channel group

taningh86, 2024-10-17:
Hi,
My recordings areas with NPX 2.0 are pretty diffeent - one is from the cortex and the other is from the hypothalamus. The electrophysiological properties of these neurons in the two regions are pretty different and in the back of my mind I have always thought that the highpass filter and CMR (common median referencing) applied to all units may not be the best way to go and may hide activity, especially in the hypothalamus. In pursuit to figure this out, I stumbled upon the 'process a recording by channel group' option that spikeinterface has. This is great and upon reading the notes it further makes sense that running high-pass filter and CMR across shanks also may not be the best way to go. 
I want to get a more expert opinion from you guys and ask how would you guys process a dataset such as mine. Would you split, preprocess, and sort channels from different brain regions and different shanks separately? If so, could you share a code to do so. 
In the tutorial https://spikeinterface.readthedocs.io/en/latest/how_to/process_by_channel_group.html , the recordings are split by individual channel. I wonder if that's really necessary and instead I can split by a group of channels, such as all channels in the cortical area of interest ? If I use splitting by groups of channels then I am assuming I will have to group them manually by identifying what channels are there in my area of interest from the probe map or is there a way in SI to find out what channels are there between specific depths of the probe? 
I have no problem with splitting, prepreprocessing and sorting each channel separately as it is there in the tutorial but just want to find out if that's better or necessary or grouping channels is another option.

Thanks!

alejoe91, 2024-10-17:
Hi @taningh86 

Actually the tutorial (written by @JoeZiminski ) splits different channel groups (i.e. shanks). Not sure where you saw the individual channel splitting :)

alejoe91, 2024-10-17:
The sorter will not make any assumption about waveform shapes, so I would suggest to juts split by shank as in the tutorial

taningh86, 2024-10-17:
Ok. I will give it a go and see what comes of it. I will update what I find and reach out if there are questions. Thanks @alejoe91

closed 2024-11-05 by alejoe91