SpikeInterface/spikeinterface #3485: Multiple template comparison 'm_tcmp'

taningh86, 2024-10-17:
Hi,
I have used pairwise 'p_tcmp' comparisons for matching units in two recording sessions and it works with the agreement scores. However, when I am comparing/matching multiple recordings `m_tcmp` I am not sure what feature to use, since m_tcmp don't take `agreement_scores`. 
My goal is to match units in all recordings at once instead of going through pairs. Doing it pair by pair, which I have been ding so far, is time intensive to match across multiple recordings. 
Please let me know how could I create something like an agreement scores that will allow me to match the similarity of unit templates across multiple recordings.

Thanks!

alejoe91, 2024-10-17:
You can use the `compare_multiple_templates` function

taningh86, 2024-10-17:
Thats what I used. But I was expecting something where units across all recordings are matched based on some score based on a template metric similar to agremment scores in pairwise comparisons. Here's the code I used 
```
import spikeinterface as si
import spikeinterface.comparison as sc

from pathlib import Path

folder1 = Path(r'G:\NPX2_2_28_24_2Rght_LHA_RSP_EXP_g0')
folder2 = Path(r'G:\NPX2_3_6_24_2Rght_LHA_RSP_just_FOR_g0')
folder3 = Path(r'G:\NPX2_24_fasted_4_19_24_EXP_g0')
folder4 = Path(r'G:\NPX2_24_fasted_4_19_24_FOR_g0')

# Load the sorting analyzer folders
analyzer_day1 = si.load_sorting_analyzer(folder1/"SA_2_28_24_EXP")
analyzer_day2 = si.load_sorting_analyzer(folder2/"SA_3_6_24_FOR")
analyzer_day3 = si.load_sorting_analyzer(folder3/"SA_4_19_24_FASTED_EXP_New")
analyzer_day4 = si.load_sorting_analyzer(folder4/"SA_4_19_24_FASTED_FOR_New")

analyzer_list = [analyzer_day1, analyzer_day2, analyzer_day3, analyzer_day4]

# match all
m_tcmp = sc.compare_multiple_templates(waveform_list=analyzer_list,
                                       name_list=["D1", "D2", "D3", "D4"])
```
                                       
With this code I get the following output, which i don't quite understand.
```
Removed node: ('D1', 5)
Removed node: ('D1', 16)
Removed node: ('D1', 39)
Removed node: ('D1', 2)
Removed node: ('D1', 13)
Removed node: ('D1', 17)
Removed node: ('D1', 7)
Removed node: ('D1', 0)
Removed node: ('D1', 9)
Removed node: ('D2', 4)
Removed node: ('D2', 40)
Removed node: ('D2', 67)
Removed node: ('D2', 6)
Removed node: ('D2', 37.0)
Removed node: ('D2', 68)
Removed node: ('D2', 2)...
```

zm711, 2024-10-18:
@alejoe91, I also don't see where we expose this to the user easily for templates. It looks like the class isn't quite as feature rich as the sorting class. it does the same graph, but then doesn't have an easy to use method for returning the agreements. Or am I wrong? 

Thanks for pointing this out Jimmy at least to me.

taningh86, 2024-10-20:
of course @zm711

taningh86, 2024-10-20:
@alejoe91 any suggestions?