SpikeInterface/spikeinterface #3486: InvalidStateError while estimating sparsity

JeffreyBoucher, 2024-10-18:
Hello all!

I've been trying to run kilosort 3 on a concatenated Neuropixels 2 dataset. Lately I've been running into an issue with create_sorting_analyzer, while it is estimating sparsity. Basically, the code is able to run about 70-80% (not any specific number) of the way through, then I get an exception "InvalidStateError". I guess this means that some aspect of my data doesn't work well with the sparsity-estimating algorithm, but I have no guess what that would be. 

Here is an example of the exception:

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/sjjgjbo/.conda/envs/neurovis_try2/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/home/sjjgjbo/.conda/envs/neurovis_try2/lib/python3.9/concurrent/futures/process.py", line 323, in run
    self.terminate_broken(cause)
  File "/home/sjjgjbo/.conda/envs/neurovis_try2/lib/python3.9/concurrent/futures/process.py", line 458, in terminate_broken
    work_item.future.set_exception(bpe)
  File "/home/sjjgjbo/.conda/envs/neurovis_try2/lib/python3.9/concurrent/futures/_base.py", line 549, in set_exception
    raise InvalidStateError('{}: {!r}'.format(self._state, self))
concurrent.futures._base.InvalidStateError: CANCELLED: <Future at 0x2aaf7896ea00 state=cancelled>

Additionally, here are the inputs into create_sorting_analyzer:

we = si.create_sorting_analyzer(recording=rec, sorting=sorting, folder=outDir / 'sortings_folder',
            format="binary_folder",
            sparse=True
            )

Please help me if you can, I would be very greatful, it's been confounding. Let me know if I can offer any additional information that would help!

Thanks,

Jeff Boucher

zm711, 2024-10-18:
I don't think any of us are using 3.9 at this point. But this is good to know. Our test suite is on 3.9 and doesn't have a problem. I'm not sure. I think we need to ping @alejoe91 and @samuelgarcia to take a look at this. 

What `n_jobs` are you using? Could you try `n_jobs=1` to confirm it is a multiprocessing issue.?

JeffreyBoucher, 2024-10-19:
Hello!

Thanks for the response! I am using n_jobs -1, which should be 10 cores on the cluster I'm using. I'll try with n_jobs = 1 and get back to you!

zm711, 2024-10-20:
Thanks let us know with the `n_jobs=1`. Sometimes there can be issues with how a server shares resources so we need to trouble shoot 3 things:
1) python 3.9 issue
2) multiprocessing issue
3) spikeinterface + server issue

JeffreyBoucher, 2024-10-21:
Hello!

Setting n_jobs = 1 indeed let me get through the sparsity estimation without error! Naturally this takes much longer to do, though.

Since you have implied I might be able to solve my problem by updating from python 3.9 I'll maybe give that a shot next. It's been a while since I chose my version, but I think having 3.9 isn't critical at this stage of the pipeline. 

Thanks for your help! I'll let you know if changing versions doesn't solve it for me; let me know if you need any more information from me. 

Jeffrey Boucher

zm711, 2024-10-21:
Yeah it would be great if you could test python 3.10 or 3.11. There have been some improvements in multiprocessing at the python level. If updating python works it tells us that 3.9 might not be as well supported as we thought for our multiprocessing. If 3.10/3.11/3.12 doesn't work then it might be a problem in our multiprocessing itself.

JeffreyBoucher, 2024-10-25:
Hello!

Unfortunately, I still got the same error with python 3.11!

:estimate_sparsity:  70%|███████   | 7983/11378 [3:11:25<36:45,  1.54it/s]
estimate_sparsity:  70%|███████   | 7990/11378 [3:11:40<1:21:16,  1.44s/it]Exception in thread Thread-2:

Traceback (most recent call last):
  File "/home/sjjgjbo/.conda/envs/spikesortEnv/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/home/sjjgjbo/.conda/envs/spikesortEnv/lib/python3.11/concurrent/futures/process.py", line 347, in run
    self.terminate_broken(cause)
  File "/home/sjjgjbo/.conda/envs/spikesortEnv/lib/python3.11/concurrent/futures/process.py", line 499, in terminate_broken
    work_item.future.set_exception(bpe)
  File "/home/sjjgjbo/.conda/envs/spikesortEnv/lib/python3.11/concurrent/futures/_base.py", line 559, in set_exception
    raise InvalidStateError('{}: {!r}'.format(self._state, self))
concurrent.futures._base.InvalidStateError: CANCELLED: <Future at 0x2b409bed9d90 state=cancelled>

Any advice on what to try next? Anything you might also want to look at?

Thanks!

Jeff Boucher

zm711, 2024-10-25:
Thanks for that info! A few more background questions then:

What OS are you using (looks like linux maybe, which flavor?)? Is this on a server or locally? If on a server what is your local OS that you are communicating with the server with?

Could you do a conda list or pip list of version numbers for your packages in the environment?

Could you give us the stats on your recording object? If you just type recording into your terminal the repr should tell us file size/dtype/number of samples?

JeffreyBoucher, 2024-10-25:
The recording I am currently working with outputs:

"
ConcatenateSegmentRecording: 384 channels - 30.0kHz - 1 segments - 341,317,801 samples 
                             11,377.26s (3.16 hours) - int16 dtype - 244.13 GiB
"

It's a set of concatenated recordings taken over a period of about a week and a half.

Thanks for your help!

Jeff Boucher

zm711, 2024-10-25:
Could you try running just one of the recordings and see if that works with `n_jobs`. I remember vaguely that we had a problem with certain concatenations so I would like to test this.

@h-mayorquin do you remember this too? That giant concatenations were causing problems with multiprocessing? 

the issue with this is the best way for us to fix this is to have the data to try to fix it with but sharing ~250 GB is a non-trivial thing :) 

Maybe @samuelgarcia or @alejoe91 also have opinions about why multiprocessing is failing with concatenation (and they both use linux!).

samuelgarcia, 2024-10-25:
Salut.
Are you running the script using slurm ?
In my lab, the slurm kill jobs because the way slurm is counting memory is wrong.
With shared mem, every process is memory is cumulated that overflow the slurm limits but the real machine limits is OK.
Can could test with less process and more thread ?
`n_jobs=6, max_thread_perprocess=8` for instance ?

JeffreyBoucher, 2024-10-28:
Hello!

I'll run a single session overnight tonight.

We are not using slurm; the cluster seems to be using "SGE 8.1.9", which stands for "Sun Grid Engine". I don't know if there would be a similar problem with this; I'll try to do a non-concatenated dataset first

JeffreyBoucher, 2024-10-30:
Hello!

In fact I ran into a bug which I think is on my end; I'm going to de-prioritize this for a bit, since I was able to get it working by turning off parallel processing I want to get that started on my real dataset, but afterward I'll get back to this (within a week)

Thanks for your help!

Jeff Boucher

samuelgarcia, 2024-10-30:
Maybe SGE is killing your job because using too much ram. Could you increase the mem when submiting the job ?

JeffreyBoucher, 2024-11-01:
Hello

Parallel processing worked fine for a single session; for that and other reasons, I think that the suggestion to request more RAM for my jobs is a good one. I'll try it!

Thanks,

Jeff

zm711, 2025-03-07:
I'll close this for now. Feel free to reopen or open new issues if anything comes up. We might need to debug other HPC/servers so happy to try helping in the future.

closed 2025-03-07 by zm711