SpikeInterface/spikeinterface #3489: How to save memory when interpolate neuropixels data more than 10min?

DaohanZhang, 2024-10-19:
Hello, I managed to interpolate a 10min's NPX 30kHz AP recording through a 30kHz rigid motion object. An 102GB array was generated, which far exceeded the computational capability of my server. What should I do?
```python
#in dredge:
me_rigid_upsampled = mu.resample_to_new_time_bins(me_rigid3, rec.get_times(0))
print(len(me_rigid_upsampled.displacement))
```
```
17820732
```
```python
from spikeinterface.sortingcomponents.motion import estimate_motion, interpolate_motion
motion_si = mu.motion_estimate_to_spikeinterface_motion(me_rigid_upsampled)
rec_interpolated = interpolate_motion(rec, motion_si)
```
```
---------------------------------------------------------------------------
MemoryError                               Traceback (most recent call last)
Cell In[41], line 3
      1 from spikeinterface.sortingcomponents.motion import estimate_motion, interpolate_motion
      2 motion_si = mu.motion_estimate_to_spikeinterface_motion(me_rigid_upsampled)
----> 3 rec_interpolated = interpolate_motion(rec, motion_si)

File ~/.conda/envs/kilosort4/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion/motion_interpolation.py:343, in InterpolateMotionRecording.__init__(self, recording, motion, border_mode, spatial_interpolation_method, sigma_um, p, num_closest, interpolation_time_bin_centers_s, interpolation_time_bin_size_s, dtype, **spatial_interpolation_kwargs)
    340 channel_inside = np.ones(locs.shape[0], dtype="bool")
    341 for segment_index in range(recording.get_num_segments()):
    342     # evaluate the positions of all channels over all time bins
--> 343     channel_displacements = motion.get_displacement_at_time_and_depth(
    344         times_s=motion.temporal_bins_s[segment_index],
    345         locations_um=locs,
    346         grid=True,
    347     )
    348     channel_locations_moved = locs[:, None] + channel_displacements
    349     # check if these remain inside of the probe

File ~/.conda/envs/kilosort4/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion/motion_utils.py:146, in Motion.get_displacement_at_time_and_depth(self, times_s, locations_um, segment_index, grid)
    143     assert times_s.ndim == 1
    144     out_shape = times_s.shape
--> 146 points = np.column_stack((times_s, locations_um))
    147 displacement = self.interpolators[segment_index](points)
    148 # reshape to grid domain shape if necessary

File ~/.conda/envs/kilosort4/lib/python3.9/site-packages/numpy/lib/shape_base.py:652, in column_stack(tup)
    650         arr = array(arr, copy=False, subok=True, ndmin=2).T
    651     arrays.append(arr)
--> 652 return _nx.concatenate(arrays, 1)

MemoryError: Unable to allocate 102. GiB for an array with shape (6843161088, 2) and data type float64
```

zm711, 2024-10-20:
@samuelgarcia

alejoe91, 2024-10-20:
I think something is wrong in the motion object. It seems you have 6843161088 temporal bins for t
10 minutes, which would correspond to a sampling frequency of the estimated motion of 11Mhz...

@cwindolf any idea what is wrong here?

samuelgarcia, 2024-10-22:
Hi,
What is the bin size on time axis for estimating the motion ?
Which method are you using to estimate the motion ?
The typical bin size is 1second. And that would make 600 bins only for 10min.

DaohanZhang, 2024-10-22:
Oh yeah! i used dredge https://github.com/evarol/dredge/issues/1, the bin size is equal to the sampling rate 30KHz.

alejoe91, 2024-10-22:
You don't need the motion sampling frequency to be 30kHz, but the recording one! When running the `interpolate_motion`, the returned recording will have the correct sampling frequency, even if the motion object is ont upsampled. Can you try to skip the upsampling motion step?

cwindolf, 2024-10-22:
Oh, sorry, I missed your screenshot above! 3.33e-05 is about 30kHz, so it looks like the motion_si is at 30kHz.

DaohanZhang, 2024-10-22:
[image]
Yes! I have used me_rigid1 (or me_rigid3, motion displacement of 50Hz), the interpolation process raised no issue. But the following issue raised in kilosort: https://github.com/evarol/dredge/issues/1
So I guess there are something wrong with the interpolation.

closed 2024-10-23 by cwindolf