SpikeInterface/spikeinterface #3494: Template loading issue and extensions documentation.

OlivierPeron, 2024-10-21:
Hi everyone,

I was recently experimenting with the analyzer extensions and found it challenging to locate the parameters for each extension. Perhaps I missed it, but I couldn't find any dedicated page or documentation listing this information. While the `get_default_extension_params` function from the `SortingAnalyzer` does provide default parameters, it doesn't offer the full range of available options.

I'm primarily raising this issue not because of that, but rather due to some problems I encountered while using templates. Similar issues might occur with other extensions, but I haven't tested those yet.

Specifically, I used 'median' as the operator to compute my templates from the waveforms, and I saved the template extension in my extension folder. The problem arises when the function `get_dense_templates_array` is called, as it assumes that templates were saved using the 'average' operator. Here's the relevant code:

```python
if isinstance(one_object, Templates):
    templates_array = one_object.get_dense_templates()
elif isinstance(one_object, SortingAnalyzer):
    if return_scaled != one_object.return_scaled:
        raise ValueError(
            f"get_dense_templates_array: return_scaled={return_scaled} is not compatible; SortingAnalyzer has the opposite setting."
        )
    ext = one_object.get_extension("templates")
    if ext is not None:
        templates_array = ext.data["average"]
    else:
        raise ValueError("SortingAnalyzer requires the 'templates' extension to retrieve templates")
else:
    raise ValueError("Input must be either Templates or SortingAnalyzer")
```

Thanks again for all the work !

PERON Olivier

alejoe91, 2024-10-21:
Hi @OlivierPeron 

I think it woould be safest to do something like this:
```
    templates_array = ext.data.get("average") or ext.data.get("median")
    assert templates_array is not None, "Average or median templates have not been computed."
```

Do you feel like trying to make a pull request?

zm711, 2024-12-04:
since a PR related to this has been merged and this appears to be solved I'll close this one :) Let us know if something else comes up!

closed 2024-12-04 by zm711