SpikeInterface/spikeinterface #3498: Method to return the original probe

h-mayorquin, 2024-10-23:
As I am taking a look I realize that the `get_probe` method returns the contact vector:

https://github.com/SpikeInterface/spikeinterface/blob/bfe9fb649b58c48d2a04949061154f75f8e08b1f/src/spikeinterface/core/baserecordingsnippets.py#L255-L282

I am looking at the `set_probe` and I don't think we save the probe as an object at all? Can I get it? Could we add something like that?

samuelgarcia, 2024-10-23:
by design we do not have the probe object attached.
The idea was to manipulate probe using the contact_vector.

alejoe91, 2024-11-05:
@h-mayorquin we store alle the probe metadata in probe info. Maybe the `get_probe` should add these metadata to the returned probe object. 

The returned object would not be the original one, but it would be a clone with proper name, manufacturer, model, contour, etc

h-mayorquin, 2024-11-05:
I think that @alejoe91 sounds like a good compromise. It should be a clone of the original probe, right?