SpikeInterface/spikeinterface #3500: feature request: drift correction computation on GPU

WeissShahaf, 2024-10-23:
Dear Devs,
the drift estimation in its different forms takes a very long time.
Is it possible to move some of the computations for drift estimation and correction to the GPU? now that torch is in spikeinterface?

Thanks.
Shahaf

alejoe91, 2024-10-23:
Hi @WeissShahaf 

Have you tried the `dredge_fast` preset? That's the fastest implementation and it should take less than half of the recording duration.

It would be possible to implement some methods with GPU acceleration, but it will take months and a dedicated project...

WeissShahaf, 2024-10-23:
i see. thanks!

samuelgarcia, 2024-10-24:
If you have a GPU "dredge_fast" which is using torch should already do the heavy computation on GPU.

alejoe91, 2024-11-05:
@WeissShahaf may we close this?

WeissShahaf, 2024-11-05:
Yes, 
Thank you