SpikeInterface/spikeinterface #3502: Kilosort docker image cant install spikeinterface

arturoptophys, 2024-10-23:
Hi all,
i have troubles to run sorting in docker containers for Kilosort 3 but also Kilosort 2.5
If i just run the sorter with docker_image=True

```
installation_mode='auto' switching to installation_mode: 'github'
Starting container
Installing spikeinterface with github in container
Installing neo with pypi in container
Running kilosort3 sorter inside spikeinterface/kilosort3-compiled-base
Stopping container
[2024-10-23 13:07:00,075][ERROR]: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "path_removed/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'
```
switching mode to pypi doesnt solve this issue.

I tried to build modified docker image with:

```
FROM spikeinterface/kilosort3-compiled-base:latest
RUN pip install --user "spikeinterface[full] @ git+https://github.com/SpikeInterface/spikeinterface.git"
```

yet here I get the error the newest spikeinterface is not compatible with python version in container.

ERROR: Package 'spikeinterface-0.102.0' requires a different Python: 3.8.1 not in '<4.0,>=3.9'

switching to installation from pypi solved the docker build, and if i point docker_image to this build it runs.

Overall, it seems the installation_mode using github is not compatible anymore with python versions in docker containers.
However, why switching  installation_mode to pypi didn't help im not sure...

Best,
Artur

zm711, 2024-10-23:
Where is the python 3.8 coming from? It went end of life in October of this year so we switched over to >=3.9 in main. Do you need python 3.8 for something?

arturoptophys, 2024-10-23:
No clue, I assumed it was from docker container. I'm running 3.10.12 locally.

zm711, 2024-10-23:
@alejoe91 knows all the docker stuff. So he can comment on this. Maybe we need to make new dockers with an appropriate python version?

samuelgarcia, 2024-10-23:
I guess we need to rebuild kilosort with more recent pyhon.
The workaround is to install spikeinterface with the 0.101.0 version which support python 3.8
now si need python > 3.8

Lingsheng-Meng, 2024-10-24:
> Same with pykilosort docker. Also fixable by creating custom docker and installing spikeinteface from pypi.

Same issue with herdingspikes docker.

alejoe91, 2024-10-28:
Thanks for pinging guys! I'll update the python3.8 images over the week!

alejoe91, 2024-10-29:
@arturoptophys I'm testing the different images and they are all Python>=3.9.

Is it possible that your local images are outdated? I would suggest removing them locally and pulling the latest image again.

- Kilosort2.5: Python 3.9.16
[image]

- Kilosort2: Python 3.9.16
[image]

- Kilosort3: Python 3.9.16
[image]

- PyKilosort: Python 3.9.17
[image]

- HerdingSpikes (just pushed v0.4.5): Python 3.12.7
[image]

arturoptophys, 2024-10-29:
Indeed, I have tried spike sorting some time ago and downloaded docker image back then.  running docker image ls showed the were build > year ago.

```
run -it spikeinterface/kilosort3-compiled-base:latest
root@41dbd15b9634:/# python --version
Python 3.8.1
```
Cant test pulling newest version right now as I run out of disc space on my pc :/ .. but I assume this would solve the issue. Sorry for the false alarm. 

Wouldn't it make sense to add "docker pull" if one is using "latest" image, when trying to spikesort? As this would prevent the issue of updating the spikeinterface repo but accessing old docker image, like happened to me.

alejoe91, 2026-03-02:
Hi @MichelleLatham 

The docker images come without a SpikeInterface installation on purpose, since they can be used as "external" images from a normal python installation. If you want to run that code standalone, you can make a new Dockerfile adding spikeinterface:
```
FROM spikeinterface/kilosort4-base:latest

RUN pip install spikeinterface==0.103.2
```