SpikeInterface/spikeinterface #3511: Channel assigned to unit problem after post processing

LuRodForster, 2024-10-28:
Hi, I am using spikeinterface and kilosort 4 to process my data. When I open phy to do my manual curation, I'm noticing that the channel indicated in the cluster view never matches the channel with the unit's waveform biggest amplitude. In this example, the best channel is 18 but I'm getting the channel n 10. It seems as if it's reporting the first channel where the unit is ever encountered instead of the best channel. This is happening with every unit in the recording and only for my exported to phy data after post processing with the spikeinterface functions. If I open the data directly from what kilosort 4 outputs then it's the correct channel. I'm not sure exactly which post processing function might be the one changing this (unit_locations, spike_locations or maybe even peak_location? Although this one I run it before the sorter) or maybe it's something else entirely?  I'm attaching the steps that I'm doing after running kilosort4 when creating the analyzer. I also compute quality metrics before exporting to phy. Thanks!
[image]
[image]
[image]

alejoe91, 2024-10-29:
@LuRodForster Could you share the KS4 output folder and the SpikeInterface-created Phy folder on google drive/dropbox, etc?

LuRodForster, 2024-10-29:
Yes, here is the [link ](https://drive.google.com/drive/folders/12Q3rM03xOLjupIsLS2JJ_F___vxlPslJ?usp=sharing)

alejoe91, 2024-10-31:
Hi @LuRodForster 

Thanks for sharing. I tried to play around with the two datasets but couldn't find a reason for the mismatch. I think it must be related to the ordering of template index, but debugging Phy is an absolute nightmare.

My recommendation would be to use SpikeInterface to unequivocally give you the extremum channel id for each unit. You can als add it as a custom column to phy as follows:

```
extremum_channel_ids = si.get_template_extremum_channel(analyzer)
analyzer.sorting.set_property("best_si_channel", list(extremum_channel_ids.values()))

# when exporting to phy
sexp.export_to_phy(..., additional_properties=["best_si_channel"])
```

LuRodForster, 2024-10-31:
Hi @alejoe91 Thanks so much for looking into this! I'll follow your recommendation and add that column

closed 2025-03-18 by zm711