SpikeInterface/spikeinterface #3512: Save spikeinterface -gui setup for other curations

Antho2422, 2024-10-29:
Hi guys, 

I just started using the spikeinterface gui to perform the curation and its great ! I was wondering if there is a way to save the setup and settings so I don't have to resetup everything each time ? Maybe @samuelgarcia knows 

Would be awesome :)

Thanks,
Anthony

Antho2422, 2024-10-29:
I also noticed that : When doing a curation and saving it as json, the resulting json does not have the version_format key. This results in an error when applying the curation using the `apply_curation()` function to the sorting analyzer

zm711, 2024-10-29:
@alejoe91 do we want to make a gui tag like we have a probeinterface tag?

alejoe91, 2024-10-29:
Yes, but this should be done on the GUI side right?

zm711, 2024-10-29:
Yeah I think so. Sam will know better, but I think the gui tag would be good for us to keep track of things we need to fix there vs here. I've got a couple labmates using the gui so I can try to dig into some of this a bit more to post for Sam.

Antho2422, 2024-10-30:
Okay thanks for the clarification good thing that it is planned