SpikeInterface/spikeinterface #3513: Spikeinterface gui issues and improvements

Antho2422, 2024-10-29:
Hello spikeinterface team, 

As mentionned here https://github.com/SpikeInterface/spikeinterface/issues/3512#issuecomment-2443927401 I am starting to use the gui to perform curation and I would like to report some issues and propose some improvements that can make the curation process a bit easier. 

I would like to mention that I'm just starting to use the gui and some of the reported issues or improvements proposed might already be in it (but I could not find it)

### Issues : 

- The first issue is the one mentionned in the referenced issue above, the fact that the resulting JSON after curation may not have been updated after the updates on the required JSON format for curation dict. This results in errors when using `apply_curation()` on the sorting analyzer
- I also noticed an error when trying to apply auto_merge. I have noticed that a huge work as been made recently on the `get_potential_auto_merge()` function. Those changes have not been applied to the gui and this result to a missmatch in expected args when using this function inside the pairlist view in the gui. Fixing this would be really helpful since the improvements in the function `get_potential_auto_merge()` seems to be good. I would love to test it into the gui

### Features and improvements:

I would also like to discuss potential improvements to the gui that can improve the user experience.

- Possibility to save the setup for next use. I mentionned this here already https://github.com/SpikeInterface/spikeinterface/issues/3512#issue-2620813474 but this could be a huge thing. 
- One thing that could be helpful (especially when dealing with merges) would be to be able to check the RPV percentage of each unit individually and also the RPV percentage of the resulting unit when applying the merge. This could help us to check visually inside the gui if the resulting unit would violate too much the refractory period or not. A plus would be to also be able to change quickly the time of the refractory period. 
One sidenote about the RPV percentage, I know that 2 metrics already exists but they are more difficult to interpret (even if they are more precise). I would like to discuss if you think it could be helpful to add a more "simple" version being just the number of spikes of the cluster that violates the RP and then divide by the total number of spikes and multiply by 100. This is a more basic that as biases but more helpful in my opinion to decide on how to perform the curation. 
Adding this metric would help. 
- Last would be to add keyboard shortcuts. I don't know if this is difficult or not I have no experience in developing such tools but since curation is a very long process, having shortcuts to assign labels quickly, change of unit with arrows, navigating in the trace views with right and left arrows would help a lot. 

I would be more than happy to discuss that with you guys, especially @samuelgarcia !

Thank you, 
Anthony

samuelgarcia, 2024-10-30:
Salut Anthony.
I will be in the gdrneuralnet meeting in 2 weeks in saclay. Will you be there ?
We can discuss theses features.

get_potential_auto_merge() is being refactoring and will be propagated in sigui very soon.

Antho2422, 2024-10-30:
Salut Samuel, 

Unfortunately I will not be there but I would love to have a meeting whenever you are free to discuss about those features !

MaximeSoula, 2025-04-15:
Hi Samuel,

That would indeed be really useful!
It would be great to have, like in Phy, the ability to split an existing cluster and save it as a new unit, for example in .npy format.

Thanks again for all the great work on the GUI!
Maxime