SpikeInterface/spikeinterface #3514: Issue in loading data (from phy)

kshtjkumar, 2024-10-29:
Hi so i created a analyser using for my data using this:

```
waveform_folder_path = r"C:\Users\garim\waveforms_kshtj_961_aug_2023_lko_230801_151701347713"
new_sorting_analyzer_path = r"C:\Users\garim\testing_phy"
wvf_extractor = si.load_waveforms(folder=waveform_folder_path)
the_recording = wvf_extractor.recording
the_sorting = wvf_extractor.sorting
channel_locations =wvf_extractor.get_channel_locations()

from spikeinterface.exporters import export_to_phy
we = si.extract_waveforms(recording=the_recording, sorting=the_sorting, folder=waveform_folder_path+str('phy'))
_  = computespike_amplitudes(waveform_extractor=we)
_  = compute_principal_components(waveform_extractor=we, n_components=3, mode='by_channel_global')
export_to_phy(waveform_extractor=we, output_folder=new_sorting_analyzer_path+str("_here_i_am"))
```
now i open the params.py file in phy and merge clusters to improve the sorting and save it. now I am trying to load the phy file after editing and saving it back in spikeinterface to generate plots and view quality matrix using this:

```
folder = r"C:\Users\garim\testing_phy_here_i_am"
wvf_extractor = si.load_waveforms(folder)
the_recording = wvf_extractor.recording
the_sorting = wvf_extractor.sorting
channel_locations =wvf_extractor.get_channel_locations()

we = si.extract_waveforms(recording=the_recording, sorting=the_sorting, folder=folder+str('re_here_is_phy_edited_wavfrm'))
  = spost.computespike_amplitudes(waveform_extractor=we)
  = spost.compute_principal_components(waveform_extractor=we, n_components=3, mode='by_channel_global')

qm  = sqm.compute_quality_metrics(waveform_extractor=we)
print(qm)
```

but it is showing the original clusters and QM and not the one that i edited manually in phy saved. I want to load the edited file that i saved from phy gui and then opened it in spikeinterface to generate plots of the wavform that were edited and merged using phy.

zm711, 2024-10-30:
@kshtjkumar why aren't you using `read_phy`. The new `Sorting` should be your phy-edited sorting, not the original `Sorting) from the waveform extractor.

So instead of using `the_sorter`. You should do 

```python
sorting = se.read_phy(waveform_folder_path + str('phy'))
```

alejoe91, 2024-11-05:
@kshtjkumar does it fix your problem? Can we close the issue?

kshtjkumar, 2024-11-05:
yes , perfect. Thank you.