SpikeInterface/spikeinterface #3515: Jupyter Notebook spikeInterface import issue HPC

htran1902, 2024-10-30:
Hi,

I use Jupyter Notebook on a HPC and I got the following error when importing spike interface by doing 
> import spikeinterface.full as si

Restarting the kernel does not change anything. Any ideas how to fix it? (I am not a Python expert) Thanks!

[image]

alejoe91, 2024-10-30:
Hi @htran1902 

Can you try to update numpy to 1.26.4?

htran1902, 2024-10-30:
The numpy version is already 1.26.4

zm711, 2024-10-30:
Could you try just doing
```bash
pip -U numba
```

htran1902, 2024-10-30:
I still have the same error (Numba version is 0.60.0) and spikeinterface version is 0.100.6

zm711, 2024-10-30:
Could you just do a full pip list so we can see what is in the environment?

htran1902, 2024-10-31:
Here's a list of modules in the environment
[image]

zm711, 2024-10-31:
The last thing you could try would be 

```
pip uninstall numpy
pip uninstall numba
pip install numba
```

this could break the environment, but basically this type of numba issue typically occurs when there is a miscommunication between numpy or numba (or both). So I would give that a go and if it still doesn't work then I think we will have to dig into how you are doing the install :)

h-mayorquin, 2024-11-01:
Besides import time, this is an extra reason to delay numba till execution.

zm711, 2024-12-03:
Any word @htran1902 ?

zm711, 2025-03-07:
Since this is 3 months stale I will close. Feel free to reopen or post a new issue if anything comes up :)

closed 2025-03-07 by zm711