SpikeInterface/spikeinterface #3521: How to eliminate potential AC noises?

DaohanZhang, 2024-11-06:
Hello everyone! I found my last NPX recording got polluted with some potential AC noises possibly due to poor GND and REF. Here are some of the noises after sorting with KS4, which is spikes approximately appeared at at 50Hz pulses. Is there any preprocess filters could eliminate such noises. I tried using notch_filter centered 50, 100 and 150Hz before default bandpass_filter, it just did not work. 
[image]
[image]

alejoe91, 2024-11-06:
Hi @DaohanZhang 

Can you plot a spectrum of the traces? If the noise is localized, you can play with the q factor of the notch filter to broaden/narrow the notch band to try to remove most of the noise

DaohanZhang, 2024-11-06:
It's a little bit weird... I have put most of the details on slack.
[image]

DaohanZhang, 2024-11-11:
Here comes the spectra!
First I plot the traces and there are periodic patterns:
[image]
After that, I plot the spectra of some of the elecs:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DaohanZhang, 2024-11-12:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
These are PSDs

zm711, 2024-12-04:
Any word on trying Alessio's tip @DaohanZhang ?

cwindolf, 2024-12-20:
Some collaborators have had good results from https://github.com/MariusKlug/zapline-plus although it is Matlab and I haven't tried it personally