SpikeInterface/spikeinterface #3547: drift correction pipeline for long / concatenated recordings  (RE: slack post)

WeissShahaf, 2024-11-21:
Dear @alejoe91 
1) for long / concatenated recordings when trying to apply drift merge (dredge_fast) in spikeinterface my RAM usage goes up to 100% (630GB). especially during peak localization. (also when only selecting some peaks). 

2) additionally after applying  recording_corrected = InterpolateMotionRecording...
and trying to save recording_corrected.save_to_folder(folder=motion_folder / 'bin' ,overwrite=True,**global_job_kwargs)
i get an error starting with (full script and error attached:
Traceback (most recent call last):
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\concurrent\futures\process.py", line 261, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\concurrent\futures\process.py", line 210, in _process_chunk
    return [fn(*args) for args in chunk]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\concurrent\futures\process.py", line 210, in <listcomp>
    return [fn(*args) for args in chunk]
            ^^^^^^^^^
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\site-packages\spikeinterface\core\job_tools.py", line 460, in function_wrapper
    return _func(segment_index, start_frame, end_frame, _worker_ctx)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\site-packages\spikeinterface\core\recording_tools.py", line 182, in _write_binary_chunk
    traces = recording.get_traces(
             ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\site-packages\spikeinterface\core\baserecording.py", line 346, in get_traces
    traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\site-packages\spikeinterface\sortingcomponents\motion\motion_interpolation.py", line 447, in get_traces
    traces = interpolate_motion_on_traces(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\weisss\micromamba\envs\SPK\Lib\site-packages\spikeinterface\sortingcomponents\motion\motion_interpolation.py", line 154, in interpolate_motion_on_traces
    drift_kernel = get_spatial_interpolation_kernel(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: get_spatial_interpolation_kernel() got an unexpected keyword argument 'temporal_bins'

[error.txt](https://github.com/user-attachments/files/17842268/error.txt)
[script.txt](https://github.com/user-attachments/files/17842269/script.txt)

alejoe91, 2024-11-21:
@WeissShahaf what SpikeInterface version are you using? I think that the second issue might be fixed in the latest release.

For the first issue, we have a way to cache detected and localized peaks to file instead of memory. This is currently not exposed as a parameter, so you should per-compute peaks and peak_locations as follows:

```
from spikeinterface.sortingcomponents.peak_detection import  

gather_mode = "npy"
gather_folder = folder
# node detect
method = detect_kwargs.pop("method", "locally_exclusive")
method_class = detect_peak_methods[method]
node0 = method_class(recording, **detect_kwargs)

node1 = ExtractDenseWaveforms(recording, parents=[node0], ms_before=0.1, ms_after=0.3)

# node detect + localize
method = localize_peaks_kwargs.pop("method", "center_of_mass")
method_class = localize_peak_methods[method]
node2 = method_class(recording, parents=[node0, node1], return_output=True, **localize_peaks_kwargs)
pipeline_nodes = [node0, node1, node2]
peaks, peak_locations = run_node_pipeline(
    recording,
    pipeline_nodes,
    job_kwargs,
    job_name="detect and localize",
    gather_mode=gather_mode,
    gather_kwargs={},
    squeeze_output=False,
    folder=gather_folder,
    names=["peaks", "peak_locations"],
)

# then you can run the estimate motion:
motion = estimate_motion(recording, peaks, peak_locations, progress_bar=progress_bar, **estimate_motion_kwargs)

# and interpolate
recording_corrected = InterpolateMotionRecording(recording, motion, **interpolate_motion_kwargs)
```
@samuelgarcia @yger this should work rihgt?

WeissShahaf, 2024-11-21:
from spikeinterface.sortingcomponents.peak_detection import  
doesn't work. what did u meant to import? or just import  *?

alejoe91, 2024-11-21:
> from spikeinterface.sortingcomponents.peak_detection import doesn't work. what did u meant to import? or just import *?
> 
> even with that ExtractDenseWaveforms is not defined

Updated imports, sorry about that

WeissShahaf, 2024-11-21:
and the kwargs are defined by the preset kwargs?

    preset_parms = get_motion_parameters_preset("dredge_fast")
    detect_kwargs=preset_parms['detect_kwargs']
    localize_peaks_kwargs=preset_parms['localize_peaks_kwargs']
    select_peaks_kwargs= preset_parms['select_kwargs']    
    estimate_motion_kwargs=preset_parms['estimate_motion_kwargs']
    interpolate_motion_kwargs = preset_parms['interpolate_motion_kwargs']

WeissShahaf, 2024-11-21:
and if i want to add a select_peaks method?

add a node between node0 and node1?
method = select_kwargs.pop("n_peaks", 10000)
    method_class = select_peak_methods[method]
    node0_5 = method_class(recording, **select_kwargs)

alejoe91, 2024-11-21:
Basically my snippet was a copy/paste of the [correct_motion](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/motion.py#L251) function, so you can use the same logic there!

WeissShahaf, 2024-11-21:
still getting an error:
node2 = method_class(recording, parents=[node0, node1], return_output=True, **localize_peaks_kwargs)
Traceback (most recent call last):

  Cell In[126], line 3
    node2 = method_class(recording, parents=[node0, node1], return_output=True, **localize_peaks_kwargs)

TypeError: LocalizeCenterOfMass.__init__() got an unexpected keyword argument 'upsampling_um'

samuelgarcia, 2024-11-22:
Hi @WeissShahaf 
also about memory issue this could come from the neo package depanding the format you are using.
Could you try to install/update with neo master branch ?

closed 2024-11-22 by samuelgarcia

alejoe91, 2024-11-22:
@samuelgarcia why did you close?

WeissShahaf, 2024-11-22:
@samuelgarcia 
did you mean to do this?
pip install git+https://github.com/NeuralEnsemble/python-neo.git
pip install git+https://github.com/SpikeInterface/probeinterface.git

and i'm using the dredge_fast parameters. only changes: 
one_preset_params = get_motion_parameters_preset("dredge_fast")
one_preset_params['detect_kwargs']['method']="locally_exclusive_torch"
one_preset_params['detect_kwargs']['noise_levels']=noise_levels
one_preset_params['select_kwargs']['method']='uniform'
one_preset_params['select_kwargs']['n_peaks']='10000'

but even with that. an 11 hour recording wasn't even done detecting peaks after 24 hours using the above suggested pipeline.
on spikeinterface version 0.101.2

samuelgarcia, 2024-11-22:
> @samuelgarcia why did you close?

by mistake i think

WeissShahaf, 2024-11-25:
FYI,
I ran into a problem when precomputing noise levels, then detect_kwargs causes an error because noise levels is already an argument: spikeinterface24\src\spikeinterface\preprocessing\motion.py  , in line 421:
 peaks = detect_peaks(recording, noise_levels=noise_levels, pipeline_nodes=None, **detect_kwargs, **job_kwargs)
i fixed that this way:
 
        if 'noise_levels' in detect_kwargs:
            peaks = detect_peaks(recording, pipeline_nodes=None, **detect_kwargs, **job_kwargs)
        else:
            peaks = detect_peaks(recording, noise_levels=noise_levels, pipeline_nodes=None, **detect_kwargs, **job_kwargs)

WeissShahaf, 2024-11-25:
also the returned object of the correcttion stage is an interpolate_motion object, not a recording. is that normal?

alejoe91, 2024-11-25:
If you precompute noise levels and pass then, you have to remove them from the detect kwargs dictionary. Alternatively you can add them to the dictionary and not pass it as argument.

The output is correct. Every recording in SpikeInterface has its own class, which inherits from the BaseRecording