SpikeInterface/spikeinterface #3549: fix handling of synchrony columns in qualitymetrics

zm711, 2024-11-22:
@chrishalcrow

I'm sorry I missed this during review, but we need to find a better way of dealing with synchrony quality metrics. I just discovered that we have the names `sync_spike_2` etc hardcoded, but those are named based on user input with 2,4,8 or 2,6,8 being the defaults. So we should think about a more robust way to deal with that. Not sure how, but I discovered this in #3497 and added a note in comment for us to think on it.

alejoe91, 2024-11-23:
Thanks @zm711 

I agree this is hard to deal with. I would lean towards hard coding the synchrony sizes...what do you think? If someone needs some customization they would run the low level function

chrishalcrow, 2024-11-27:
Hello, sorry for the delay. I agree that we should just hard code the synchrony sizes: it's gonna be a lot of pain for an edge case. I'll have a go in the next couple of days?

zm711, 2024-11-27:
That would be amazing from my perspective. I'm super busy the next week (and it is a holiday tomorrow in the US) so I definitely won't have time to change.

PR #3559 (2024-12-05): Hard code `synchony_sizes`
Fix #3549

Currently there is one metric, `synchony`, whose column names depends on a kwarg: `synchony_sizes`. If the user inputs `synchony_sizes = [3,5,7]` the columns in the `quality_metrics` dataframe are `sync_spikes_3`, `sync_spikes_5` and `sync_spikes_7`. This is a bit awkward to deal with in a few places. Computing these custom `synchrony_sizes` are a bit of an edge case, so me @alejoe91 and @zm711 decided to just hard code the default values of `synchony_sizes = [2,4,8]`.

To help users who _really_ want to compute custom synchony sizes, I've left the `_compu [... cut]