SpikeInterface/spikeinterface #3551: Issue with Highpass Filtering

majidkhalili, 2024-11-25:
Hello everyone,

I’m encountering unexpected behavior when applying a highpass filter to my data. After filtering, all meaningful information seems to be lost.

To investigate, I manually filtered a segment of the signal and compared it to the results from the spikeinterface filtering function. However, the outputs are dramatically different.

Below is the code for filtering with spikeinterface and using scipy:

```
# Spikeinterface filtering
recording_HF = sif.highpass_filter(recording, freq_min=300)
ch133 = recording.get_traces(start_frame=5440*30000, end_frame=5460*30000, channel_ids=['AP133'])
ch133_filt_spikeinterface = recording_HF.get_traces(start_frame=5440*30000, end_frame=5460*30000, channel_ids=['AP133'])

# Manual filtering
lowcut = 300  
fs = 30000  
order = 4  

b, a = scipy.signal.butter(order, lowcut / (fs / 2), btype='low')
ch133_filt_scipy = scipy.signal.filtfilt(b, a, ch133.T)
```

And here is the result:

[image]

Environment Details:

> Recording: Neuropixels
> spikeinterface version: I tried both SpikeInterface 0.101.2 and SpikeInterface 0.101.0
> Python version: 3.11
> Operating system: Windows

I would appreciate any insights or suggestions on how to resolve this discrepancy. Could this be due to differences in filter implementation or parameter settings? Could you please also provide insights into best practices for applying highpass filters in this context? 

Thank you

alejoe91, 2024-11-25:
Hi, in your example you're applying a low-pass filter. For spike sorting, the interesting information is in the high frequency!

majidkhalili, 2024-11-25:
OMG, it has been bugging me for several hours today, I have oversighted the the filtering parameter in the btype paramer in the butter function. Thanks a lot

closed 2024-11-25 by majidkhalili