SpikeInterface/spikeinterface #3555: computing amplitude cutoff

karmenrai2, 2024-11-28:
I am trying to computer the amplitude cutoff using this code but I keep just getting nan for every value. Is there something wrong with the code or is it a data issue?

`# Load data using SpikeInterface
recording = si.read_binary(file_paths=rec_file_path, sampling_frequency=sampling_frequency,
                           num_channels=num_channels, dtype=dtype)
sorting = read_phy(sort_file_path)

# Load data using numpy
all_labels = np.load(cluster_label)
all_pcs = np.load(pcs)

# Set the probe
other_probe = get_probe('cambridgeneurotech', 'ASSY-37-E-1')
other_probe.set_device_channel_indices(np.arange(32))
recording = recording.set_probe(other_probe, group_mode='by_shank')

# Create sorting analyzer
analyzer = si.create_sorting_analyzer(sorting=sorting, recording=recording, format="memory")
if __name__ == '__main__':    
    analyzer.compute("random_spikes", method="uniform", max_spikes_per_unit=600, seed=2205)
    analyzer.compute("waveforms", ms_before=1.3, ms_after=2.6, n_jobs=2)
    analyzer.compute("templates", operators=["average", "median", "std"])
    analyzer.compute(input="spike_amplitudes")

# Quality Metrics
presence_ratio = sqm.compute_presence_ratios(sorting_analyzer=analyzer)
amplitude_cutoff = sqm.compute_amplitude_cutoffs(sorting_analyzer=analyzer, peak_sign="both")

# CSV file 
with open('spikesorting.csv', 'w') as file:
    writer = csv.writer(file)
    writer.writerow(["Cluster", "Presence Ratio", "Amplitude Cutoff", "Isolation Distance"])
    for cluster in presence_ratio.keys():
        isolation_distance, _ = sqm.mahalanobis_metrics(all_pcs=all_pcs, all_labels=all_labels, this_unit_id=cluster)
        writer.writerow([cluster, presence_ratio[cluster], amplitude_cutoff.get(cluster, ""), isolation_distance])

# Print data for verification
print(recording)
print(sorting)
print(analyzer)
print(presence_ratio)
print(amplitude_cutoff)`

here is the warning received during runtime
`warnings.warn("compute_amplitude_cutoffs need 'spike_amplitudes' or 'waveforms' extension")`

this is the output I am getting

Cluster | Presence Ratio | Amplitude Cutoff | Isolation Distance
-- | -- | -- | --
  |   |   |  
0 | 1 | nan | 1.89105441986244
  |   |   |  
1 | 1 | nan | 21.9754870562538
  |   |   |  
2 | 1 | nan | 2.08639005321337
  |   |   |  
3 | 1 | nan | 1.42021378063038
  |   |   |  
4 | 1 | nan | 0.614645809462376
  |   |   |  
5 | 1 | nan | 0.209511255264518
  |   |   |  
6 | 1 | nan | 0.548317966330339
  |   |   |  
7 | 1 | nan | 0.492024220726647
  |   |   |  
8 | 1 | nan | 0.0672568742271259
  |   |   |  
9 | 1 | nan | 2.09064727504505
  |   |   |  
10 | 1 | nan | 0.64246598552092
  |   |   |  
11 | 1 | nan | 1.75604923652422

alejoe91, 2024-11-28:
How long is your recording? The amplitude cutoff needs a minimum number of spikes.to estimate distributions

zm711, 2024-12-02:
Yeah for what it's worth for my stuff we do recordings of relatively slow spiking neurons for 1-3 hours. So I basically always get `nans` as well. Like Alessio said if your recording conditions (or your neurons) are not amenable to distribution estimation than it won't work. I just never rely on this since so much of my data doesn't work with it.

closed 2024-12-03 by karmenrai2