SpikeInterface/spikeinterface #3558: Spyking-circus no clusters and no spikes

dornierv, 2024-11-29:
Hello,

I'm writing to you because I'm encountering this problem for the first time. I have run the spyking-circus algorithm via spikeinterface for several patients without problems, but for this particular dataset I have encountered some problems. 

I'm working with tetrodes acquired with a neuralynx system and you can see below the signal from two tetrodes on the same probe. 
As you can see from the screenshot, there are spikes for the first tetrode (top) but after running the spyking-circus algorithm, there is no cluster assigned to this tetrode and spikes are not detected. On the other hand, for the second tetrode (bottom), spikes are detected and assigned to different clusters.

I was wondering if you had an idea on how to resolve this, I tried different things but none of them worked (e.g., change the threshold, adjust a dead file manually to ignore all periods with artefacts, run the algorithm for the first tetrode only).

Thank you in advance for your time and your answer,
Best regards,

[image]

Vincent Dornier

zm711, 2024-12-02:
@yger

yger, 2024-12-03:
are we talking about SC1 or SC2 ?

dornierv, 2024-12-03:
This is with SC1

zm711, 2025-04-04:
SC1 is not actively maintained any more. You could test SC2 as that is the direction moving forward. I'll close this for inactivity, but feel free open a new issue if needed :)

closed 2025-04-04 by zm711