SpikeInterface/spikeinterface #3561: scale parameter in kilosort4

YangLin916, 2024-12-02:
Hi! I am using kilosort4 in spikeinterface to do spike sorting. I notice the scale parameter in kilosort4. Should I set it to the same as gain_to_uV property in my recording data, or just leave it to None? My recording data is in binary format, and acquired from our lab's own system. And I manually set the gain_to_uV property to the data. 

Another question: I found running kilosort4 in spikeinterface is slower than running it separately through KS4 GUI. I am curious why it is.

Thanks a lot!

zm711, 2024-12-03:
> Hi! I am using kilosort4 in spikeinterface to do spike sorting. I notice the scale parameter in kilosort4. Should I set it to the same as gain_to_uV property in my recording data, or just leave it [...]

Could you explain this a bit more. You are just doing `si.read_binary` from your proprietary format? Is there a header? I think I need at least need more info to understand this. :) 

> Another question: I found running kilosort4 in spikeinterface is slower than running it separately through KS4 GUI. I am curious why it is.

We have a wrapper around KS4. Adding extra layers of python will always slow things down. Could you quantify how much slower? We often find the biggest slow-down for people who want to use shank isolation in SpikeInterface. KS4 will try to automatically do this for the templates with dminx, which allows it to pass through the data less, then if you actually isolate the data with our wrapper which would then go through the data 1 time per shank. We think the hit in speed is worth the increase in accuracy. But small slowdowns are absolutely expected when wrapping other code. :)

YangLin916, 2024-12-17:
Thank you for your reply!

For the first point: I load the data by `recording = se.read_binary(file_paths = base_folder / 'NPElectrophysiology_2_2.bin', sampling_frequency=30000, num_channels=385, dtype = "int16", gain_to_uV=2.3438, offset_to_uV=0)`
And I want to know if I need to change this`'scale': 'Scaling factor to apply to data before all other operations. Default None.' 
` params in KS4 sorter.

For the second one, I ran a 3-hour recording collected with Neuropixels NHP in the Kilosort4 GUI, and it took 5.5 hours. However, running it through the KS4 wrapper in SpikeInterface took 7 hours. I used the default params in both of them.

zm711, 2025-01-24:
Sorry for the delay, the holiday break dropped this off our radar. For scale I would directly ask the KS4 team to understand what the need for that is! And I think  5.5 vs 7 hours could be reasonable within the context of a wrapper. It's what about 1.2 or 1.3x slowdown which makes sense!

zm711, 2025-02-05:
I'll close this for now. Hopefully you get more info from the KS4 team on the scale parameter :)

closed 2025-02-05 by zm711