SpikeInterface/spikeinterface #3562: Does using kilosort3 in docker require GPU access now?

taningh86, 2024-12-02:
Hi,
Wondering if recent updates in Spikeinterface or docker have changed setting such that it requires GPU access now. While running kilosort3 I am getting 'Unable to find a supported GPU device' error, which I think is docker trying to access GPU and failing because pytorch is not currently installed in my 'si_env' environment. 
Please confirm if that is the case or there's more to this. If GPU access is now required then I am going to install Pytorch in my environment compatible with CUDA 11.8.
Thanks
Jimmy

alejoe91, 2024-12-02:
Hi, Kilosort3 has always required a GPU. The docker image comes with per-installed CUDA binaries, but the GPU needs to be available on the system. Does running`nvidia-smi` work?

taningh86, 2024-12-02:
yeah running nvidia-smi works and gives the follwing info as usual:```

Mon Dec  2 09:46:39 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 556.12                 Driver Version: 556.12         CUDA Version: 12.5     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                  Driver-Model | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  Quadro RTX 5000              WDDM  |   00000000:01:00.0  On |                  Off |
| 34%   32C    P8             15W /  230W |    5403MiB /  16384MiB |      7%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1560    C+G   C:\Users\Gregg\anaconda3\python.exe         N/A      |
|    0   N/A  N/A      8556    C+G   ...__8wekyb3d8bbwe\Notepad\Notepad.exe      N/A      |
|    0   N/A  N/A     12440    C+G   ...oogle\Chrome\Application\chrome.exe      N/A      |
|    0   N/A  N/A     13128    C+G   C:\Users\Gregg\anaconda3\python.exe         N/A      |
|    0   N/A  N/A     13916    C+G   ...2txyewy\StartMenuExperienceHost.exe      N/A      |
|    0   N/A  N/A     13936    C+G   C:\Windows\explorer.exe                     N/A      |
|    0   N/A  N/A     14284    C+G   ...nt.CBS_cw5n1h2txyewy\SearchHost.exe      N/A      |
|    0   N/A  N/A     14452    C+G   ...\Docker\frontend\Docker Desktop.exe      N/A      |
|    0   N/A  N/A     16960    C+G   ...CBS_cw5n1h2txyewy\TextInputHost.exe      N/A      |
|    0   N/A  N/A     17572    C+G   ...oogle\Chrome\Application\chrome.exe      N/A      |
|    0   N/A  N/A     19176    C+G   ...crosoft\Edge\Application\msedge.exe      N/A      |
|    0   N/A  N/A     19512    C+G   ...es (x86)\Dropbox\Client\Dropbox.exe      N/A      |
|    0   N/A  N/A     20568    C+G   ...\Local\slack\app-4.40.133\slack.exe      N/A      |
|    0   N/A  N/A     21616    C+G   ...__8wekyb3d8bbwe\WindowsTerminal.exe      N/A      |
|    0   N/A  N/A     23976    C+G   ...5n1h2txyewy\ShellExperienceHost.exe      N/A      |
|    0   N/A  N/A     24660    C+G   ...on\HEX\Creative Cloud UI Helper.exe      N/A      |
|    0   N/A  N/A     24720    C+G   C:\Users\Gregg\anaconda3\python.exe         N/A      |
|    0   N/A  N/A     27304    C+G   C:\Users\Gregg\anaconda3\python.exe         N/A      |
|    0   N/A  N/A     27764    C+G   ...es (x86)\Dropbox\Client\Dropbox.exe      N/A      |
|    0   N/A  N/A     28724    C+G   ...t Office\root\Office16\POWERPNT.EXE      N/A      |
+-----------------------------------------------------------------------------------------+
```

alejoe91, 2024-12-02:
Mmm, then there might be some incompatibility between your CUDA version and the one in the docker image...Can you share the entire error you're getting?

alejoe91, 2024-12-02:
Thanks. The error is not due to a missing GPU, but some internal Kilosort3 failure... It could be something wrong with your data or some high levels of localized activity/noise which makes whitening failing

taningh86, 2024-12-02:
hmmm...ok. I thought the whitening also had to do something with the GPU issue. What is the GPU not found error then? 
Also, I have ran the same data multiple times and did not use to get the same error. The only thing is that I have not used SI for few months. I am using the 101.0 version of SI and was wondering if anything have changed since my last use.
What do you recommend for this issue?

alejoe91, 2024-12-02:
Ahh sorry I misread the error.. yeah it could be an incompatibility issue between the GPU-CUDA version on your system and the one KS was compiled against. Unfortunately, in this case the only solution would be to compile a new image with the required versions...

alejoe91, 2024-12-02:
We don't specify any notebook version (also there are many different packages involved). If you have problems with Jupyter and spikeinterface, please open another issue about it!

zm711, 2025-01-14:
We will close this one for now, but like Alessio said open more issues if they come up. From what our group has seen though, I will say, that we've had issues with different versions of Kilosort playing poorly with different versions of matlab so I think most people in the lab are using relatively old matlab since that was what the older kilosorts were written against (but then you have to worry about newer GPUs with older Matlab and CUDA versioning so it can be even more annoying).

closed 2025-01-14 by zm711