SpikeInterface/spikeinterface #3565: time_bin_centers_s is a list

BEGINRX, 2024-12-03:
I met a problem.

```
Traceback (most recent call last):
  File "/home/NITD/lxr/code/spike_sorting/run_sorter1013.py", line 450, in start_sorting
    sorting = run_sorter_ms5(rec_w_sub, 
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/lxr/code/spike_sorting/run_sorter1013.py", line 272, in run_sorter_ms5
    sort_res = si.run_sorter('mountainsort5', recording, 
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py", line 199, in run_sorter
    return run_sorter_local(**common_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py", line 261, in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 302, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 262, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/external/mountainsort5.py", line 124, in _run_from_folder
    recording = cls.load_recording_from_folder(sorter_output_folder.parent, with_warnings=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 215, in load_recording_from_folder
    recording = load_extractor(pickle_file, base_folder=output_folder)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1196, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 776, in load
    d = pickle.load(f)
        ^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 529, in from_dict
    extractor = _load_extractor_from_dict(dictionary)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1135, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sortingcomponents/motion/motion_interpolation.py", line 390, in __init__
    interpolation_time_bin_centers_s, interpolation_time_bin_edges_s = ensure_time_bins(
                                                                       ^^^^^^^^^^^^^^^^^
  File "/home/NITD/anaconda3/envs/si_env_rolling/lib/python3.11/site-packages/spikeinterface/sortingcomponents/motion/motion_utils.py", line 607, in ensure_time_bins
    time_bin_edges_s = np.empty(time_bin_centers_s.shape[0] + 1, dtype=time_bin_centers_s.dtype)
                                ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'list' object has no attribute 'shape'
```

I don't know what I should do. Help me. Thank you.

zm711, 2024-12-03:
Could you post your script. It looks like you are running mountainsort5, but then I see motion correction as well. I need to figure out where the motion stuff is coming from.

I think based on the error we fixed this in main. Which version of spikeinterface are you using? :)

BEGINRX, 2024-12-04:
I had update the version. Now the version shown is 0.102.0.
[image]
The code is following:
```
    sort_res = si.run_sorter('mountainsort5', recording, 
                             scheme=scheme,
                             detect_threshold=detect_threshold,
                             detect_sign=detect_sign,
                             snippet_T1=snippet_T1,
                             snippet_T2=snippet_T2,
                             scheme2_phase1_detect_channel_radius=phase1_detect_channel_radius,
                             detect_time_radius_msec=detect_time_radius_msec,
                             npca_per_channel=phase1_npca_per_channel,
                             npca_per_subdivision=phase1_npca_per_subdivision,
                             filter=False,
                             whiten=False,
                             remove_existing_folder=True, 
                             delete_output_folder=True,
                             verbose=False, 
                             output_folder=output_folder
                            )
```
I want to update to the version 0.101.2, but after using the rolling script, the version is 0.102.0...Dose the version cause the porblem?

BEGINRX, 2024-12-04:
I removed the enviorment, and downloaded the new one. It seems run well. I think it's the version problem.

closed 2024-12-04 by BEGINRX