SpikeInterface/spikeinterface #3566: Default `plot_unit_locations` method is not what is documented

chrishalcrow, 2024-12-03:
Hello,

Our docstring for `compute_unit_locations` says:
 ```
method : "center_of_mass" | "monopolar_triangulation" | "grid_convolution", default: "center_of_mass"
```

But I think the default is actually `monopolar_triangulation`, as set here https://github.com/SpikeInterface/spikeinterface/blob/853d8a48abcf40f5a47401ed2f7baa83e1c3295f/src/spikeinterface/postprocessing/unit_locations.py#L56

I think we want to use `center_of_mass` by default, as is done for `compute_spike_locations`. @samuelgarcia @alejoe91 can you confirm?

zm711, 2024-12-03:
I'm actually curious about these thought processes. We are seeing some interesting jumping around as discussed with @cwindolf somewhere (sorry I forget where we discussed "weird" template shapes causing units to show up in really weird places). In general I like monopolar except for edge cases (which for me are usually actually artifact). Why do we want a COM default vs monopolar_traingulation. (I haven't tests the grid_conv to know if I like it...)

chrishalcrow, 2024-12-04:
I have no strong opinion about which default we use, let's pick one and document it correctly 👍

zm711, 2024-12-04:
@alejoe91 was your idea for COM to do the computationally easy thing as default and then choose the more intense stuff by user-choice. I guess I can see the argument in that if that's the reason we start with COM. (or is it just the commonness of using COM in general? )

zm711, 2024-12-04:
So I think at least Alessio and I agree then. I like the current defaults and I think we should just fix the docs to fit with the current defaults. Unless someone has a vote against this.

chrishalcrow, 2024-12-05:
Sounds good - made a PR

PR #3570 (2024-12-05): Correct method default in docstring
Fixes #3566

Correct the default `method` docstring in `compute_unit_locations`. Was `center_of_mass` but `monopolar_triangulation` is actually the default.
closed 2024-12-05 by alejoe91