SpikeInterface/spikeinterface #3572: Different channel maps in one acquisition

atombysx, 2024-12-07:
I have Neuropixel 2.0 acquisitions with each segments having different channel maps (spikeGLX) to check activities across shanks with stimuli. But it seems like it only provides one channel map for one recording object (loading first segment's channel map).

I want to take out segments with different channel maps before feeding to kilosort by run_sorter()

Is there any function that can be used to load different channel maps from segments? Or it's best to learn how to feed the desired channel map to the recording object before sorting?

atombysx, 2024-12-12:
For anyone who is interested:
Seems like read_spikeglx() only reads the first g file's channel map and does not support such customization. But probeinterface has a read_spikeglx() function to read .meta files to load the probes. So my solution is to use si.read_spikeglx() to load the acquisition and select segments with the same channel maps and create new recording objects. Then use probeinterface to load the correct channel maps from the meta files (or you can also use read_imro() to load channel maps directly but .meta files can't be wrong). Set the new recordings with the new probe channel maps:

```
import matplotlib.pyplot as plt
import spikeinterface.full as si
import probeinterface as pi

#load the recording
recording = si.load_extractor('/data_path/recording.bin')

#load the probe layout and plot it
recording.get_probe().to_dataframe()
fig, ax = plt.subplots(figsize=(15, 10))
si.plot_probe_map(recording, ax=ax)

#load other metafile to get another probe layout
meta_path = 'recording_path/20241126_0_g1_imec0/20241126_0_g1_t0.imec0.ap.meta'
new_probe = pi.read_spikeglx(meta_path)
#read_imro() works as well

#select segments 1 and 2 to create a new recording and change the probe layout
segments=[1,2]
new_recording = si.select_segment_recording(recording,segment_indices=segments)
new_recording.set_probe(new_probe,in_place=True)

#plot the new probe layout
fig, ax = plt.subplots(figsize=(15, 10))
si.plot_probe_map(new_recording, ax=ax)
```

closed 2024-12-12 by atombysx

samuelgarcia, 2025-01-06:
Hi @atombysx 
I am not sure to be awre of this use case.
You have a spikeglx meta file which contain several segments with differents channel maps ?
I was thinking that a meta file was a unique channel maps.

samuelgarcia, 2025-01-08:
OK thanks.