SpikeInterface/spikeinterface #3573: Recent update breaks hdsort in the container.

rat-h, 2024-12-09:
I recently updated `spikeinterface`.
```python
Python 3.10.12 (main, Nov  6 2024, 20:22:13) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import spikeinterface.full as si
>>> si.__version__
'0.101.2'
```
A script which uses containerized `hdsort` now returns an error:
```python
2024-12-09 17:48:56,804:871   ERROR   :Sorting failed: Sorting failed: Spike sorting in singularity failed with the following error:
Traceback (most recent call last):
  File "/home/rth/spikeinterface/20240321_PV_P8A_PSAM4_Thamus_rec_Thalamus-truncated-hdsort-local-T2.50-20241021-115820-20240321_PV_P8A_PSAM4-HDS-Lx32x32x6-4106399-final-final-165/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'

Traceback (most recent call last):
  File "/home/rth/spikeinterface/runthepipe3.py", line 220, in runsorter
    sorting = si.run_sorter(
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 193, in run_sorter
    return run_sorter_container(
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 648, in run_sorter_container
    raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in singularity failed with the following error:
Traceback (most recent call last):
  File "/home/rth/spikeinterface/20240321_PV_P8A_PSAM4_Thamus_rec_Thalamus-truncated-hdsort-local-T2.50-20241021-115820-20240321_PV_P8A_PSAM4-HDS-Lx32x32x6-4106399-final-final-165/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/rth/spikeinterface/runthepipe3.py", line 869, in <module>
    sorting_saved = runsorter(last,preproc_saved)
  File "/home/rth/spikeinterface/runthepipe3.py", line 228, in runsorter
    raise RuntimeError(f"Sorting failed: {e}")
RuntimeError: Sorting failed: Spike sorting in singularity failed with the following error:
Traceback (most recent call last):
  File "/home/rth/spikeinterface/20240321_PV_P8A_PSAM4_Thamus_rec_Thalamus-truncated-hdsort-local-T2.50-20241021-115820-20240321_PV_P8A_PSAM4-HDS-Lx32x32x6-4106399-final-final-165/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/rth/spikeinterface/runthepipe3.py", line 872, in <module>
    raise RuntimeError(f'Sorting failed: {e}')
RuntimeError: Sorting failed: Sorting failed: Spike sorting in singularity failed with the following error:
Traceback (most recent call last):
  File "/home/rth/spikeinterface/20240321_PV_P8A_PSAM4_Thamus_rec_Thalamus-truncated-hdsort-local-T2.50-20241021-115820-20240321_PV_P8A_PSAM4-HDS-Lx32x32x6-4106399-final-final-165/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load_extractor
ModuleNotFoundError: No module named 'spikeinterface'
```

I rolled back to the 0.101.0 version and everything works well. I've tried version 0.101.1, and it returns the same error as version 0.101.2.

alejoe91, 2024-12-09:
Hi @rat-h 

I think that the container is based on python 3.8 which is not supported anymore. Do you have a particular reason to run hd sort? It's not maintained anymore and it gives sub optimal performance. I would lean towards moving hd sort to the legacy unsupported sorters.

alejoe91, 2024-12-10:
You can run hd sort without any container if you have Matlab installed! :)

rat-h, 2024-12-10:
Dear @alejoe91,
Are you suggesting that I should switch from `spikeinterface` and python to Matlab? WoW! (0.0)

alejoe91, 2024-12-10:
No. You can run any Matlab based sorter within spikeinterface if you have Matlab installed and an environment variable set. You can check the docs!

rat-h, 2024-12-12:
@alejoe91, silly question.
why
```python
os.environ["HDSORT_PATH"] = "/home/spikeinterface/.local/apps/HDsort
sorting = si.run_sorter(  sorter_name="hdsort" ....
```
does not work, but
```bash
HDSORT_PATH="/home/spikeinterface/.local/apps/HDsort" python myscript.py
```
does?

Is it possible to set path within python script? I haven't tried `os.system('export HDSORT_PATH="/home/spikeinterface/.local/apps/HDsort" ')`, but it doesn't look like a good approach in anyway.

alejoe91, 2024-12-12:
The env variable must be available prior to importing spikeinterface. I recommend adding the export to the bashrc

samuelgarcia, 2024-12-13:
You can force in older version of spikeinterface inside the docker. Find the most recent one that support py3.8 and it should work.

@alejoe91  : maybe we should have a machinery to install a very minimal `spikeinterface.sorter` API like just to run inside the docker and that should py3.8 (and even before, compatible).
What do yout think ? The host would keep the latest spikeinterface but in the container we should able to run older version. No ?

And also, you can update the actual container image to install more recent python.

rat-h, 2025-04-14:
Please, everyone, we are stuck with SI version 0.101.0, the last version when `hdsort` worked in the container. If you want to have a voice from your user (i.e., from me :), it would be awesome if containers had a minimal SI and fixed Python version so they wouldn't be broken. **The ability to support many sorters (also pretty old ones), which run in the container, was a brilliant idea and an amazing feature.** Unfortunately, this feature slowly decays with versions. First, `yass` was moved to 'retrospective support', then `hdsort.` Can this fantastic feature of the spikeinterface shine again?