SpikeInterface/spikeinterface #3574: can't change parameters of sorting

dmkobylkov, 2024-12-11:
Hi!

I am trying to sort with MS5 my tetrode data (16 tetrodes). The sorting runs fine, but when I want to change the sorting scheme, it seem to ignore the changes. Here is the code I am using:

```
default_ms5_params = sorters.Mountainsort5Sorter.default_params()
default_ms5_params['scheme'] = '3'
print(default_ms5_params)

folder_name = "Sort_Tetrodes"  # Customize folder naming
folder_path = os.path.join(datapath, folder_name)
aggregate_sorting = sorters.run_sorter_by_property(sorter_name='mountainsort5', recording=recording_neural,
                                           grouping_property='group',
                                           folder=folder_path, **default_ms5_params)

```
Here is a part of the output I see, and it still SCHEME2 written there. Could you, please, tell me, why is it the case?

```
{'scheme': '3', 'detect_threshold': 5.5, 'detect_sign': -1, 'detect_time_radius_msec': 0.5, 'snippet_T1': 20, 'snippet_T2': 20, 'npca_per_channel': 3, 'npca_per_subdivision': 10, 'snippet_mask_radius': 250, 'scheme1_detect_channel_radius': 150, 'scheme2_phase1_detect_channel_radius': 200, 'scheme2_detect_channel_radius': 50, 'scheme2_max_num_snippets_per_training_batch': 200, 'scheme2_training_duration_sec': 300, 'scheme2_training_recording_sampling_mode': 'uniform', 'scheme3_block_duration_sec': 1800, 'freq_min': 300, 'freq_max': 6000, 'filter': True, 'whiten': True, 'delete_temporary_recording': True, 'n_jobs': 4, 'chunk_duration': '1s', 'progress_bar': True, 'mp_context': None, 'max_threads_per_process': 1}
write_binary_recording 
n_jobs=4 - samples_per_chunk=30,000 - chunk_memory=468.75 KiB - total_memory=1.83 MiB - chunk_duration=1.00s
=============================================
Processing block 1 of 2...
Using training recording of duration 300 sec with the sampling mode uniform
*** MS5 Elapsed time for SCHEME2 get_sampled_recording_for_training: 0.227 seconds ***
Running phase 1 sorting
Number of channels: 4
Number of timepoints: 9000000
Sampling frequency: 30000.0 Hz
Channel 0: [-1.8369702e-15 -1.0000000e+01]
Channel 1: [-1.0000000e+01  1.2246468e-15]
Channel 2: [6.123234e-16 1.000000e+01]
Channel 3: [10.  0.]
Loading traces
*** MS5 Elapsed time for load_traces: 0.000 seconds ***
Detecting spikes
Adjacency for detect spikes with channel radius 200
[[0, 1, 2, 3], [0, 1, 2, 3], [0, 1, 2, 3], [0, 1, 2, 3]]
```

alejoe91, 2024-12-11:
The spikeinterface code looks correct..@magland can you check on your end?

dmkobylkov, 2024-12-11:
Thanks, @magland !

closed 2024-12-11 by dmkobylkov