SpikeInterface/spikeinterface #3581: `get_unit_adaptive_window` leads to recursion errors if there are no peaks.

zm711, 2024-12-16:
In the adaptive window code it tries to find the `win_size` by taking the initial threshold set by the user and if that doesn't work then it divides by 2 to try a smaller threshold and continues. But if there is no threshold then we end up with reaching the maximum recursion depth because 0/2 will keep being 0. I don't know this bit of the code well enough to know what our base condition should be in the case that there are no peaks (maybe if a unit with bad correlograms?) so I'm no completely sure how to fix this. What do think @yger since you've updated this extensively. 

function below for convenient check:
https://github.com/SpikeInterface/spikeinterface/blob/6fde997424c589679b70158be2dcb33144594f31/src/spikeinterface/curation/auto_merge.py#L717-L755

samuelgarcia, 2025-01-06:
@yger this is clearly for you.

zm711, 2025-01-24:
@yger, just want to circle around. Now that Sam is showing off the auto_merge function this recursion error is more likely to come up for labs that have units that have few peaks, so we should find a solution.

yger, 2025-01-24:
Sorry for the having missed this one. I'll have a look, but I never saw a crash there.

zm711, 2025-01-24:
I had it happen. I think the issue is we have some neurons that fire extremely rarely so the threshold drops to 0. But thanks for bragging about how amazing your data always is :P

zm711, 2025-03-07:
Fixed by #3487.

closed 2025-03-07 by zm711