SpikeInterface/spikeinterface #3589: Kilosort4 Spike Sorting Value Error

ajcheng1, 2024-12-18:
I was running kilosort4 and it worked for one file, but when I tried a different dataset, I get this error: 
```python
  File ".local/lib/python3.11/site-packages/kilosort/run_kilosort.py", line 648, in detect_spikes
    raise ValueError('No spikes detected, cannot continue sorting.')
``` 
I think it's related to the fact the data is low density recordings with 16 electrodes. Is there a way to still use kilosort4 to spike sort? 

```python
recording = se.read_binary(dat_file_path, dtype='float32', num_channels=16, sampling_frequency=12500)
print("Original recording:", recording)
recording_f = si.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording_cmr = si.common_reference(recording_f, reference='global', operator='median')
recording_preprocessed = recording_cmr.save(format="binary")

kilosort_params = {
    'min_template_size': 4,  # Reduced from default 10
    'template_sizes': 3,     # Reduced from default 5
    'nearest_chans': 8,      # Reduced from default 10
    'nearest_templates': 50, # Reduced from default 100
    'n_templates': 4,        # Reduced from default 6
    'n_pcs': 3,             # Reduced from default 6
    'whitening_range': 16,   # Reduced from default 32
    'Th_universal': 7,       # Reduced from default 9
    'Th_learned': 6,        # Reduced from default 8
    'do_correction': False,  # Disable drift correction for low channel count
    'batch_size': 30000,    # Reduced from default 60000
    'cluster_pcs': 32       # Reduced from default 64
}

# Run Kilosort with optimized parameters
sorting_KS4 = ss.run_sorter(
    sorter_name="kilosort4",
    recording=recording_preprocessed,
    folder=output_folder,
    remove_existing_folder=True,
    **kilosort_params
)
print("Sorting complete:", sorting_KS4)

```

borrepp, 2024-12-18:
Hi, 

I'm taking the initiative to share how we have used KS4 with Plexon probes (16 channels). If your probe has similar specs, I hope this information might help you. 

First, we define some general parameters from the probe:
```python

# Number of channels in the probe
nChans = 16

#The median of the vertical contact spacing in microns
step_chan = 100.0

# Number of nearest channels to consider when finding local maxima during spike detection. We use 2 for channel spacing of 100 microns, change it to cover a radius between 100 to 200 microns
nearest_chans = 1 # or 2

# Whitened Threshold (STD from Absolute traces)
detect_threshold = 3

# Snippet params:
# the number of ms before the event to include in the snippet
ms_before = 0.6
# the number of timepoints after the event to include in the snippet
ms_after =  1.5

snippet_T1 = int(numpy.ceil(ms_before * sampling_frequency / 1000.0))
snippet_T2 = int(numpy.ceil(ms_after * sampling_frequency / 1000.0))
nt =  snippet_T1 + snippet_T2

```

But first, note that you should whiten your data :

After applying Common median referencing:

```python
import spikeinterface.preprocessing as spre
recording_whiten = spre.whiten(recording_cmr, mode='local', radius_um = step_chan*3, dtype="float32")
recording_preprocessed = recording_withen.save(format="binary")
```

With these variables, here are the parameters that have worked relatively decent for us

```python
kilosort_params = {
      'Th_universal': 8, # Default 9 # To detect more units overall, it may help to reduce Th_universal.
      'Th_learned': 9, # Default 8 # If few spikes are detected, or if you see neurons disappearing and reappearing over time when viewing results in Phy, it may help to decrease Th_learned
      'Th_single_ch': detect_threshold, # Whitened Threshold (STD from Absolute traces)
      'nt': nt, # Default 61 for 30K
      'nt0min': snippet_T1, # default 20 for 30K
      'dmin': step_chan, # A good value for dmin based on the median distance between contacts, Default None
      'dminx': 1,  # Default 32 for Neuropixels (two columns ar at 32 microns). Try setting this to the median lateral distance between contacts to start
      'min_template_size': 10, # Default 10  Standard deviation of the smallest, spatial envelope Gaussian use for universal templates for Neuropixels ([X-horizontal = 32, Y-vertical = 15|20] channel spacing in microns)
      'nearest_chans': nearest_chans, # Number of nearest channels to consider when finding local maxima during spike detection. Default: 10
      'max_channel_distance': step_chan, # Templates farther away than this from their nearest channel (i.e., the closest channel assigned to the template) will not be used. Default None = max(dmin, dminx)
      'nearest_templates': min(100, nChans), # Default 100, less than or equal to the number of channels helps avoid numerical instability
      'x_centers': 1, # Number of x-positions to use when determining center points for template groupings, Default = None
      'do_CAR': False,
      'do_correction': False,
      'skip_kilosort_preprocessing': True,
      'whitening_range': nearest_chans,
      'keep_good_only': True, 
      'save_extra_vars': True,
      'torch_device': 'cuda'
}
```

Hope this information helps
Best
Pepe

zm711, 2024-12-19:
Thanks @borrepp, I would also suggest posting this issue on the Kilosort Repo to ask them what they think. I know in general they recommend higher density probes in general, so if @borrepp 's workaround helps you then you can give it a go, but this is less a SpikeInterface level issue and more a question for Kilosort's team about optimizing. Does that make sense from the SpikeInterface side?

ajcheng1, 2024-12-19:
Thanks @borrepp! Currently don't have access to several GPUs to run it, but I'll definitely try it.

closed 2024-12-19 by ajcheng1

ajcheng1, 2024-12-27:
Happy to say after many trial and error, finally got it to work on my probe! Really appreciated all the help and I didn't have to use kilosort2.
Thought I might share my code which is a slight variation of @borrepp in case anyone else has to same issue down the road:
```python
# Additional parameters for Kilosort4 configuration
# The median of the vertical contact spacing in microns
step_chan = 350.0

nearest_chans = 1  # Number of nearest channels to consider
detect_threshold = 3  # Whitened Threshold (STD from Absolute traces)

# Snippet parameters
ms_before = 0.6  # ms before the event to include in the snippet
ms_after = 1.5  # ms after the event to include in the snippet
sampling_frequency = recording.get_sampling_frequency()

# Convert milliseconds to timepoints
snippet_T1 = int(np.ceil(ms_before * sampling_frequency / 1000.0))
snippet_T2 = int(np.ceil(ms_after * sampling_frequency / 1000.0))
nt = snippet_T1 + snippet_T2
# Preprocess recording
print("\nPreprocessing recording...")
recording_f = si.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording_cmr = si.common_reference(recording_f, reference='global', operator='median')
recording_whiten = spre.whiten(recording_cmr, mode='local', radius_um = step_chan*3, dtype="float32")
recording_preprocessed = recording_cmr.save(format="binary")
print("\nRunning Kilosort...")
kilosort_params = {
    # Use default thresholds first as recommended, change only if necessary
    'Th_universal': 9,         
    'Th_learned': 8,
    'Th_single_ch': detect_threshold, 
        
    # Basic recording parameters
    'nt': nt,                 # Keep your existing time window
    'nt0min': snippet_T1,     # Keep your existing alignment point
        
    # For 16 channels, use 2-4x channels (32-64) rounded to multiple of 32, lower runs faster
    'nearest_templates': 32,   # 
        
    # Don't modify spatial parameters unnecessarily
    'dminx': 350,            # 
    'dmin': None,             # Let it use defaults
    'max_channel_distance': None,  # Remove - no maximum in space needed
        
    # Keep default template parameters
    'min_template_size': 10,  # Use default
    'nearest_chans': 1,     # Use default
    'x_centers': 1,         # Use default
        
    # Processing settings
    'do_CAR': False,           # Common average reference
    'do_correction': False,     # Keep drift correction off since MEA plate (diff for neuropixel)
    'skip_kilosort_preprocessing': True,  # Since you do your own
        
    # GPU settings
    'save_extra_vars': True    # Save debug info
}

sorting_KS4 = ss.run_sorter(
    sorter_name="kilosort4",
    recording=recording_preprocessed,
    folder=output_folder,
    remove_existing_folder=True,
    **kilosort_params
)
print("Sorting complete:", sorting_KS4)

```

samuelgarcia, 2025-01-06:
Hi.
When you have no spike with a sorter a good check is to use the detect_peaks method we have implemented and check how many peaks (=spike without cluster label) you have in your traces.
It can help.
Notes that this function implements several methods you can play with.