SpikeInterface/spikeinterface #3590: problem with Klusta in docker container

paul-aparicio, 2024-12-18:
Hello, I was trying to run Klusta in a container using singularity and received the following error. I am not sure why the error trace seems to be importing Combinato? Below the error is the record from the singularity import. Thank you in advance for any help!

> SpikeSortingError: Spike sorting in singularity failed with the following error:
> Traceback (most recent call last):
>   File "/media/paul/Test/TY20210401_640_signalCheck_morning/processed/results_ms5/in_container_sorter_script.py", line 5, in <module>
>     from spikeinterface.sorters import run_sorter_local
>   File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/__init__.py", line 2, in <module>
>     from .sorterlist import *
>   File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/sorterlist.py", line 1, in <module>
>     from .combinato import CombinatoSorter
>   File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/combinato/__init__.py", line 1, in <module>
>     from .combinato import CombinatoSorter
>   File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/combinato/combinato.py", line 10, in <module>
>     from spikeinterface.extractors import CombinatoSortingExtractor
>   File "/usr/local/lib/python3.7/site-packages/spikeinterface/extractors/__init__.py", line 4, in <module>
>     from .bids import read_bids_folder
>   File "/usr/local/lib/python3.7/site-packages/spikeinterface/extractors/bids.py", line 4, in <module>
>     import pandas as pd
> ModuleNotFoundError: No module named 'pandas'

importing image terminal messages:

> INFO:    Creating SIF file...
> klusta-base.sif
> Starting container
> Installing spikeinterface with folder in container
> Installing neo with pypi in container
> Running klusta sorter inside spikeinterface/klusta-base
> Stopping container

alejoe91, 2024-12-18:
@paul-aparicio what spikeinterface version are you using? 

For recent-ish versions, klusta is not supported anymore because it needs python 3.7, and it is not supported anymore on the SI side.

paul-aparicio, 2024-12-18:
@alejoe91 Thanks for your response, I am using SI 0.102. so that is the issue. I was hoping that running klusta with docker would bypass the older python issue!

alejoe91, 2024-12-18:
No, unfortunately we had to drop support... :( I'll close the issue!

closed 2024-12-18 by alejoe91