SpikeInterface/spikeinterface #3591: GPU usage by Kilosort4 when running with run_sorter_by_property

jazlynntan, 2024-12-19:
Hello,

I'm running kilosort4 for a single shank using run_sorter_by_property(). Using Kilosort4 independently in the same conda environment, the same data (with all 4 shanks) took about 1.5h. However, a single shank within spikeinterface took about 6h. This leads me to suspect that the GPU is not being used?

This is the output while kilosort within spikeinterface was running:
[image]

The GPU memory seems to be used by the process but the speed seems to suggest that the GPU is not used for computation. Meanwhile the CPU usage appeared to be maxed out.

This is the code I'm using:
```
sorted = si.run_sorter_by_property('kilosort4',
                          shank1,
                          grouping_property='group',
                          folder=os.path.join('shank1_output'),
                          verbose=True,
                          engine="joblib",engine_kwargs={"n_jobs": 16},**params_kilosort4)
```

I tried using 'auto' and 'cuda' for the torch_device parameter, but both faced the same issue.

May I know if I am doing something wrong? Thank you!

JoeZiminski, 2024-12-19:
Hi @jazlynntan I have no idea and this is just a quick guess, but I'm not sure what will happen regarding GPU access when  when parallelising multiple sortings over separate cores. Presumably the separate processes are all attempting to compute on the GPU but from the runtime it doesn't seem like they are sequentially accessing the GPU in any useful way. It might be worth testing by running with `n_jobs=1` and see if this results in the GPU being used.

zm711, 2024-12-19:
That was Sam's idea too in another issue (I forget which one). He recommended doing `engine='loop'` instead so that it occurs serially when n_jobs>1 rather than all jobs trying to access the gpu at the same time.

zm711, 2024-12-19:
Also we might need to add a note to our docs to explain that joblib might not play well with gpu-based sorters. Not sure, but this is the second issue related to this.

zm711, 2025-01-06:
Yep just let us know if loop solves it. Might take longer to do though!

zm711, 2025-01-24:
Hey @jazlynntan any word on whether loop allowed you to use this?

jazlynntan, 2025-01-24:
Hi, sorry for the delay. Using loop made no difference, but when I tried on a different computer, the joblib option worked - there was GPU usage and the speed was reasonable. While I didn't dig into it, I have a slight suspicion that it could be the OS (the windows PC worked but not the linux?) or the permissions I had on the computer.

zm711, 2025-01-24:
That's good to know. Although if you were on different computers we would need to know the size of your GPU (and it's dedicated RAM) as well as your machine's RAM. That would help us think about OS issue vs resource difference. But it is valuable to know that looping still didn't work. which KS4 version are you using?

zm711, 2025-02-21:
I'll close for now but if you run into the issue again in a reproducible way let us know so we can try to debug more.

closed 2025-02-21 by zm711

harshk95, 2025-05-27:
Hi, I am having the same issue with KS4 taking 16h or so to run on an NPX 2.0 probe recording of ~1 hour with si.run_sorter_by_property(...).
I have: 
spikeinterface v0.102.4
kilosort v4.0.37
torch for cuda 11.8 
I see the same thing with GPU memory being used but the process runs very slowly. I have engine set to 'loop' and torch device set to 'cuda'.
I am running it on Windows 10 Enterprise, with an NVIDIA TITAN RTX and 24Gb memory. During the run it appears ~8Gb RAM.

Any pointers on what could be a way to troubleshoot this? I am going to try on another machine as above in the thread, but since I saw this issue thought I would add here anyway