SpikeInterface/spikeinterface #3592: Question: loading error for NP1016 probes

meszena, 2024-12-20:
This has probably been done before, but it hasn't been solved for me yet, so I would like to ask you a short question: I am trying out a new human NP dataset recorded with the newest NP1016 version (same as NP1015 - basically a "2 column 1.0", just sterile, so it received a novel serial number), and I got this error below. I searched for it here and on the NP Slack channel and I saw other people had the same issue. You suggested installing python-neo and probeinterface. I did that but I am still getting this error. What could be a fix? Thanks in advance!
Domi

[image]

closed 2024-12-20 by meszena

samuelgarcia, 2025-01-06:
Apparently we have https://github.com/SpikeInterface/probeinterface/blob/main/src/probeinterface/io.py#L1090
do you have the latest version of probeinterface ?

meszena, 2025-01-06:
Thank you, I think I possibly ran into some version issues, yes!
And now it's working fine, so we can close this!
Thanks again!
Domokos